f1xpl/openauto #97: Pre-built packages

j-ak-e, 2018-06-14:
Is it possible to pre-build the packages such that an end-user doesn't have to spend the several hours required to compile the project from source?

I'm currently looking into debian packaging, but it's a little above my skill level

WarDrake, 2018-06-14:
You might want to look into : https://github.com/opencardev/crankshaft

j-ak-e, 2018-06-14:
I have done, unfortunately for some reason Crankshaft doesn't work on my first gen model B, it sits at the crankshaft splash screen and goes no further. I'm currently waiting for OpenAuto to compile from source, at which point I can start poking it to find out what's up

KLVN, 2018-06-14:
I'm also using a first gen model B (overclocked to 950 MHz) and it took me ~7 hours to compile everything.

j-ak-e, 2018-06-14:
@KLVN This is why I suggested pre-compiling packages, it's definitely something that can be done, and would streamline the process significantly

I'm not using an OC, because I don't have a heatsink

frostedshrapnel, 2018-06-29:
This would be great.  Crankshaft is cool, but is a very stripped down OS install.  I found myself stuck between dual booting it via Betty boot and having all sorts of issues because of it, or installing only it on an SD and missing out on other things the pi is good at (emulation, ECU control).  The compromise was compiling open auto on a fresh raspbian image.  This would be much easier if in package form however not sure how much work that would be to maintain.