f1xpl/openauto #98: Error when building openauto.

ylee-one, 2018-06-17:
NOOB alert!
I started by building aasdk everything seemed to work. Then moved on to open auto no problems until step 7. I thought it might have to do with file name and or placement. I see **Note: If needed, adjust paths accordingly to your aasdk and aasdk_build directories location.** but not sure how to do that, if that is the issue. Below is LXT info from last working step to error:

pi@raspberrypi:~/openauto_build $ cmake -DCMAKE_BUILD_TYPE=Release -DRPI3_BUILD=TRUE- DAASDK_INCLUDE_DIRS="/home/pi/include"- DAASDK_LIBRARIES="/home/pi/aasdk/lib/libaasdk.so"- DASSDK_PROTO_INCLUDE_DIRS="/home/pi/aasdk_build"- DAASDK_PROTO_LIBRARIES="/home/pi/aasdk/lib/libaasdk_proto.so" ../openauto
-- Boost version: 1.62.0
-- Found the following Boost libraries:
--   system
--   log
--   unit_test_framework
--   date_time
--   log_setup
--   filesystem
--   thread
--   regex
--   chrono
--   atomic
-- Found libusb-1.0:
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib/arm-linux-gnueabihf/libusb-1.0.so
-- Found rtaudio:
--  - Includes: /usr/include/rtaudio
--  - Libraries: /usr/lib/arm-linux-gnueabihf/librtaudio.so
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/openauto_build
pi@raspberrypi:~/openauto_build $ make
[  2%] Automatic moc, uic and rcc for target btservice
[  2%] Built target btservice_automoc
[ 14%] Built target btservice
[ 17%] Automatic moc, uic and rcc for target autoapp
[ 17%] Built target autoapp_automoc
[ 19%] Building CXX object CMakeFiles/autoapp.dir/src/autoapp/App.cpp.o
/home/pi/openauto/src/autoapp/App.cpp:20:40: fatal error: f1x/aasdk/USB/AOAPDevice.hpp: No such file or directory
 #include <f1x/aasdk/USB/AOAPDevice.hpp>
                                        ^
compilation terminated.
CMakeFiles/autoapp.dir/build.make:62: recipe for target 'CMakeFiles/autoapp.dir/src/autoapp/App.cpp.o' failed
make[2]: *** [CMakeFiles/autoapp.dir/src/autoapp/App.cpp.o] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/autoapp.dir/all' failed
make[1]: *** [CMakeFiles/autoapp.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
pi@raspberrypi:~/openauto_build $

When I change file folder location in the individual files it seems to advance to a different file with the same error or file not found. Hence why I think it is file a location problem. What am I doing wrong? NOOB reminder.

rofelice, 2018-06-21:
Double check your Includes Directory
DAASDK_INCLUDE_DIRS="/home/pi/include"

Mine was /home/pi/aasdk/include

ylee-one, 2018-06-22:
That worked. Thanks!

closed 2018-06-22 by ylee-one

Bobet21, 2018-09-21:
Mine was direct from the root /aasdk/include