f1xpl/openauto #100: Stuck at Waiting for Device

rofelice, 2018-06-21:
Hello,

I have followed all of the instructions for installation and setup.  
The builds all succeed and the udev rules have been set. 
I have my USB Configuration set to MTP.
I am running Android 8.0.0
I have openauto set to autostart

When i plug in my phone Android Auto launches, but the openauto screen remains at the Waiting for Device screen.

I have reviewed the Closed Issues for similar issues and it appears that I have already covered all the "fixes"
 
Any thoughts?

jppm99, 2018-06-21:
I'm having exactly the same issue.
Already tried to re-flash the system and did it all again but with no luck

jppm99, 2018-06-21:
Thanks man.
After hours of searching everything I could find related to this issue (yes, I'm new to this) you saved me.
It worked perfectly after doing that.

rofelice, 2018-06-22:
@WarDrake this ended up working for one of my devices, but not the other.  Since I'm very much over the whole process at this point I'm going to close this issue and go with the one working device.  Thanks for the tip.

closed 2018-06-22 by rofelice