f1xpl/openauto #103: Stuck in waiting for device screen

Rufcca, 2018-06-26:
I followed all the steps of the wiki, but i get this message when i start OpenAuto

MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
[2018-06-26 07:30:18.602024] [0x724bd000] [warning] [OpenAuto] [RecentAddressesList] failed to read configuration file: openauto_wifi_recent.ini, error: openauto_wifi_recent.ini: cannot open file. Empty list will be used.
[2018-06-26 07:30:18.638631] [0x6fab7310] [info]    [OpenAuto] [App] Waiting for device...
[2018-06-26 07:30:18.642282] [0x6fab7310] [info]    [OpenAuto] [App] Devices enumeration result: false
[2018-06-26 07:30:26.443331] [0x6fab7310] [info]    [OpenAuto] [App] Device connected.
[2018-06-26 07:30:26.459912] [0x724bd000] [debug]   [OpenAuto] [AudioInput] create.
PulseAudioService: pa_context_connect() failed

my phone says that it is connected to the car, but my RPI stays in the initial screen.

WarDrake, 2018-06-26:
That's a problem with your audio config, to confirm switch audio from rt to qt on settings and connect the phone if it connects, then the likely culprit is the echo cancelation if you did that step, if not look for what else you can undo from your audio changes.

Rufcca, 2018-06-26:
i switched from rt to qt and nothing changed, i'm still stuck in the waiting screen

WarDrake, 2018-06-27:
Then use "pulseaudiuo -vvvv" to debug your pulseaudio setup, because the error : 

> PulseAudioService: pa_context_connect() failed

 means the pulse daemon is not running, and that's what causes openAuto to freeze while the phone is connected.

use the pi user while debugging, do not do it as root since the dameon runs as pi.

Rufcca, 2018-06-28:
sorry for the delay, i used the command as u said and got this 
I: [pulseaudio] main.c: setrlimit(RLIMIT_NICE, (31, 31)) failed: Operation not permitted
D: [pulseaudio] core-rtclock.c: Timer slack is set to 50 us.
D: [pulseaudio] core-util.c: RealtimeKit worked.
I: [pulseaudio] core-util.c: Successfully gained nice level -11.
I: [pulseaudio] main.c: This is PulseAudio 10.0
D: [pulseaudio] main.c: Compilation host: arm-unknown-linux-gnueabihf
D: [pulseaudio] main.c: Compilation CFLAGS: -g -O2 -fdebug-prefix-map=/build/pulseaudio-M85hft/pulseaudio-10.0=. -fstack-protector-strong -Wformat -Werror=format-security -Wall -W -Wextra -pipe -Wno-long-long -Wno-overlength-strings -Wunsafe-loop-optimizations -Wundef -Wformat=2 -Wlogical-op -Wsign-compare -Wformat-security -Wmissing-include-dirs -Wformat-nonliteral -Wpointer-arith -Winit-self -Wdeclaration-after-statement -Wfloat-equal -Wmissing-prototypes -Wredundant-decls -Wmissing-declarations -Wmissing-noreturn -Wshadow -Wendif-labels -Wcast-align -Wstrict-aliasing -Wwrite-strings -Wno-unused-parameter -ffast-math -fno-common -fdiagnostics-show-option -fdiagnostics-color=auto
D: [pulseaudio] main.c: Running on host: Linux armv7l 4.14.34-v7+ #1110 SMP Mon Apr 16 15:18:51 BST 2018
D: [pulseaudio] main.c: Found 4 CPUs.
I: [pulseaudio] main.c: Page size is 4096 bytes
D: [pulseaudio] main.c: Compiled with Valgrind support: no
D: [pulseaudio] main.c: Running in valgrind mode: no
D: [pulseaudio] main.c: Running in VM: no
D: [pulseaudio] main.c: Optimized build: yes
D: [pulseaudio] main.c: FASTPATH defined, only fast path asserts disabled.
I: [pulseaudio] main.c: Machine ID is ab5b28d043f34d788dacdeb9d7acc8ea.
I: [pulseaudio] main.c: Session ID is c1.
I: [pulseaudio] main.c: Using runtime directory /run/user/1000/pulse.
I: [pulseaudio] main.c: Using state directory /home/pi/.pulse.
I: [pulseaudio] main.c: Using modules directory /usr/lib/pulse-10.0/modules.
I: [pulseaudio] main.c: Running in system mode: no
E: [pulseaudio] pid.c: Daemon already running.
E: [pulseaudio] main.c: pa_pid_file_create() failed.

i searched for a solution but none that i found worked, do you know what is going on ?

f1xpl, 2018-06-30:
Do you have microphone connected or USB sound card with microphone output? It looks like no.

Rufcca, 2018-07-02:
i don't, it is mandatory for the app to work ?

f1xpl, 2018-07-03:
According to your pulseaudio output it is needed to connect microphone input.

closed 2018-07-03 by f1xpl