f1xpl/openauto #106: Buttons are blank

Sonnet71, 2018-06-29:
Today I switched my installation from the micro SD card to ab USB stick. Meaning I‘ve setup everything new 🙁

Mostly all went well, but one thing  is left. When I boot up the Raspberry Pi Openauto starts immidiatly. But on the cover screen all buttons have a white background with white text. Haven’t seen this before. Any advice how to change this?
Cheers
Alex

f1xpl, 2018-07-03:
What desktop driver did you use (looks like graphic artifacts)?

Lerxst151, 2018-07-05:
I noticed this a few times also.... If it auto loaded it got the white boxes.  If I exited out and just restarted OA then it looked fine.  Before I got a chance to really dig into it I messed up something else and did a new reload and everything seems to be fine now.   I have noticed over the past month or so that colors and such seem to be off on the 7" screen in OA and just on the Pi desktop.  I tried increasing the GPU memory but same issue.  I guess I'm using whatever driver that is the default on Raspian.

Sonnet71, 2018-07-05:
I‘m just using the the stock Raspbian (Version June 2018) driver. The only thing I changed was using the newest release of Openauto. 

I did than a roll back to the April Version of Raspian an the previous release of Openauto and all is fine again

DanielXYZ2000, 2018-07-07:
Hi, I have the same issue, but the text is readable if you click on it (and hold the button) - the text is very bright

Curvin777, 2018-07-11:
Chiming in with another "same here" post.

quattrofoto, 2018-07-19:
I get something similar - if Openauto loads on start up then everything is fine, although for some reason it cuts off the end of 'device' (it says "Waiting for devic").  If I close out of Openauto for whatever reason, and then reload it by typing /home/pi/openauto/bin/autoapp then when it reloads it has the white buttons described by others so you can't read their legends.  Weirdly though, when this happens it doesn't cut the "Waiting for device" message.

Beebock, 2018-08-05:
Adding a 'me too' I can restart as much as I want, the buttons are still blank

Using Latest Rasbian and compiled against last code today

whiznishant, 2018-08-24:
I too faced same issue of blank buttons. Running with sudo fixes it for me. Without sudo button are white and white text over it is barely visible, but running with sudo buttons shows up as grey and text is visible.

CluelessGenius, 2018-12-13:
@whiznishant noticed the same on my own but how do i sudo and have it autostart. manually sudo fixes the problem but autostarting with LDXE in my case gives me blank buttons again
im super new to linux please forgive me if im just being stupid