f1xpl/openauto #107: Waiting for device screen

azntklc, 2018-07-02:
Complete noob here, this is my first project and would like some assistance.

I´m currently unable to go past the ¨waiting for device screen¨ with the phone plugged in
I was reading in the closed issues section as well but didn´t seem to find anything related with the error messages that I got. I´m running this on Pi3+. My phone (note 8) says its connected to my vehicle but open auto has no response. I´ve changed my audio settings from rt to qt like other threads have suggested, didn´t work. Tried with in line sound card plugged in, no response. At this point I´m not even sure if I´m heading the right direction.

This is the message log I get everytime I run the openauto command but I´m not sure if this is related to why I´m stuck in ¨waiting for device screen¨..

libEGL warning: DRI2: failed to authenticate

** (autoapp:1442): WARNING **: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
[2018-07-01 22:45:33.061334] [0x723f7000] [warning] [OpenAuto] [RecentAddressesList] failed to read configuration file: openauto_wifi_recent.ini, error: openauto_wifi_recent.ini: cannot open file. Empty list will be used.
[2018-07-01 22:45:33.107004] [0x6f9ff310] [info]    [OpenAuto] [App] Waiting for device...
[2018-07-01 22:45:33.108057] [0x6f9ff310] [info]    [OpenAuto] [App] Devices enumeration result: false

Any ideas?

WarDrake, 2018-07-02:
Neither one of those errors is an issue, the a11y.Bus is an accesibility warning you'll get if you run any GUI without a couple exrta packages, you can safely ignore it.

The wifi_recent error is there until you use a wificonnection and it saves the file, you'd have better luck posting a complete log for us to see.

azntklc, 2018-07-02:
What command do I use to obtain the log files?

Tried /var/log/messages and I got a permission denied message

This is what I got for using dmesg command.

**[ 1843.819699] usb 1-1.2: 1:1: cannot set freq 48000 to ep 0x1
[ 1843.819868] usb 1-1.2: 2:1: cannot set freq 48000 to ep 0x82
[ 1843.820049] usb 1-1.2: 13:0: cannot get min/max values for control 2 (id 13)
[ 1843.820105] usb 1-1.2: 9:0: cannot get min/max values for control 2 (id 9)
[ 1843.820174] usb 1-1.2: 10:0: cannot get min/max values for control 2 (id 10)**

Could those USB logs be an issue?

I also found a lot of under voltage codes, I wonder if it could simply be that. Iḿ going to see if I have a high amp rating power source for the pi in the meantime. 

Thanks

WarDrake, 2018-07-02:
run the app from a terminal with the output routed to a file
"/user/pi/openauto/bin/autoapp > /home/pi/openauto.log" 
make sure the path is correct I don't have mine handy to check is that's the real path, that file will be your log.

azntklc, 2018-07-02:
So I ran the app from a terminal with ¨/home/pi/openauto/bin/autoapp¨ exited the app cause I can´t get back into terminal without doing so. Then ran ¨/home/pi/openauto.log¨ got message saying ¨no such file or directory¨. Looked through the path under openauto directory, I can´t find a .log txt document in that folder. I did find one in openauto_build folder but it only contains what was shown to me when I previously ran the command dmesg earlier.

WarDrake, 2018-07-02:
You have to run the whole command, "/home/pi/openauto/bin/autoapp > /home/pi/openauto.log" as one line, the > tells the program to route all it's output to the file on the right, run it as a single command and the file will be created.

azntklc, 2018-07-03:
Ok, this is what i got,

[2018-07-02 22:00:41.491371] [0x72478000] [warning] [OpenAuto] [Configuration] failed to read configuration file: openauto.ini, error: openauto.ini: cannot open file. Using default configuration.
[2018-07-02 22:00:41.588027] [0x72478000] [warning] [OpenAuto] [RecentAddressesList] failed to read configuration file: openauto_wifi_recent.ini, error: openauto_wifi_recent.ini: cannot open file. Empty list will be used.
[2018-07-02 22:00:41.620915] [0x6faff310] [info]    [OpenAuto] [App] Waiting for device...
[2018-07-02 22:00:41.624649] [0x6faff310] [info]    [OpenAuto] [App] Devices enumeration result: false

f1xpl, 2018-07-03:
Looks like problem related to usb setup. Did you configure udev rules correctly?

azntklc, 2018-07-04:
Configured udev rules and now it works! Thanks to the both of you. How I just have to figure out how to get my USB in line sound card working. :D

closed 2018-07-04 by f1xpl