f1xpl/openauto #108: DSO missing from command line

schuest, 2018-07-03:
Hello there,

I followed the build instructions for the openauto and everything works. But on Step 7. I got the following error:

`[  4%] Linking CXX executable /home/pi/openauto/bin/btservice
/usr/bin/ld: CMakeFiles/btservice.dir/btservice_automoc.cpp.o: undefined reference to symbol '_ZTVN10__cxxabiv121__vmi_class_type_infoE@@CXXABI_1.3'
//usr/lib/arm-linux-gnueabihf/libstdc++.so.6: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
CMakeFiles/btservice.dir/build.make:192: die Regel für Ziel „/home/pi/openauto/bin/btservice“ scheiterte
make[2]: *** [/home/pi/openauto/bin/btservice] Fehler 1
CMakeFiles/Makefile2:67: die Regel für Ziel „CMakeFiles/btservice.dir/all“ scheiterte
make[1]: *** [CMakeFiles/btservice.dir/all] Fehler 2
Makefile:83: die Regel für Ziel „all“ scheiterte
make: *** [all] Fehler 2
`

I know i want to try i

f1xpl, 2018-07-03:
Please post console output in english.

KLVN, 2018-07-04:
Translated:
`
[ 4%] Linking CXX executable /home/pi/openauto/bin/btservice /usr/bin/ld: CMakeFiles/btservice.dir/btservice_automoc.cpp.o: undefined reference to symbol '_ZTVN10__cxxabiv121__vmi_class_type_infoE@@CXXABI_1.3' //usr/lib/arm-linux-gnueabihf/libstdc++.so.6: error adding symbols: DSO missing from command line collect2: error: ld returned 1 exit status CMakeFiles/btservice.dir/build.make:192: recipe for target „/home/pi/openauto/bin/btservice“ failed make[2]: *** [/home/pi/openauto/bin/btservice] Error 1 CMakeFiles/Makefile2:67: recipe for target „CMakeFiles/btservice.dir/all“ failed make[1]: *** [CMakeFiles/btservice.dir/all] Error 2 Makefile:83: recipe for target „all“ failed make: *** [all] Error 2
`

f1xpl, 2018-07-06:
What Raspbian version do you use? Please provide more details.

DanielXYZ2000, 2018-07-07:
I had a similar error - what command did you start from the command line? maybe a typo?

KLVN, 2018-07-09:
@DanielXYZ2000 Probably `make`, as this error occurs if something's wrong in the Makefile.

schuest, 2018-07-10:
sorry for the late answer.

I used KODI as platform and there I get these errors.
I tried it with a fresh install of raspbian stretch and there occored no errors.

f1xpl, 2018-07-10:
Next please provide more details about used platform.

closed 2018-07-10 by f1xpl