f1xpl/openauto #111: App crashes after swiping up

newbiechi, 2018-07-08:
Relative newbie - Everything seems fine on the RPi/OpenAuto end after connecting via usb.  OpenAuto detects the connection and the AndroidAuto screen is shown.  On the phone, AndroidAuto app launches but when I swipe up, the app crashes.

Has anyone experienced something similar?

DGxInfinitY, 2018-07-08:
Yes actually. Your speaking about on the phone right?

newbiechi, 2018-07-08:
Correct, on the phone. I checked the USB settings on the phone and set permissions on the Rpi, but still no dice.

f1xpl, 2018-07-10:
If crash is related to the Android Auto app, I suppose that Android Auto helpdesk is better place to find support in investigation of the Android Auto app crash.

If you think that issue is related to the OpenAuto, please provide ADB logs.

closed 2018-07-14 by f1xpl