f1xpl/openauto #113: Support for DAB via Welle.io APP and USB Stick (Realtek RTL2832U)

Bengelator, 2018-07-08:
Hello,
is it possible to add Support for DAB via the Welle.io App when the  DAB USB Stick (Realtek RTL2832U) is connected to the Raspbery Pi?

The Welle.io App is supported by Android Auto but requires the connection via USB to the Phone (OTG)
wich then leaves no place for the usb connection to the Raspberry.

Regards
Bengelator

WarDrake, 2018-07-08:
Might work with a powered OTG cable, did you try that?

Bengelator, 2018-07-08:
Thanks for the reply. Yes it works with the OTG cable. Phone <- (OTG Calble) Usb <- USB DAB Stick

but how do i connect the raspberry Pi then?. I Could use a Usbhub and then connect the usb cable to the raspberry but does Android Auto use a OTG connection?

WarDrake, 2018-07-08:
use a "powered" OTG cable, it has 2 input ports

f1xpl, 2018-09-11:
OpenAuto Pro (new project that I am involved in) supports adding and managing of 3rd party applications. Take a look at the https://www.youtube.com/watch?v=9sTOMI1qTiA to see if this is something you are looking for.

inksnnov, 2018-09-11:
OpenAuto Pro support steering wheel buttons via gpio?(Volume, ch+-,mode,voice comand,answert)?