f1xpl/openauto #115: whatsapp call control (answer/decline) not getting routed to openauto on rpi

protonicjo, 2018-07-11:
I am using an external bluetooth adapter and followed the steps to setup HFP on the rpi. In openauto settings, i set it to use exteranl bluetooth adapter and gave the adapter address. before i connect my phone, i paired my phone to the rpi bluetooth adapter using call and media profile. Still when i receive a whatsapp call, i dont see the controls popup for answering or declining. If that is currently not supported, is it in the pipeline to enable it? My current car console does that for all voip/app based calls. And since openauto will ideally replace it, shouldnt that be a supported feature? Thanks.

f1xpl, 2018-07-11:
OpenAuto is just an Android Auto headunit emulator - it is not a complex solution to replace existing head unit. If Android Auto does not support whatsapp calls (as far as I know AA supports only standard dialer) then OpenAuto will not support it as well.

protonicjo, 2018-07-11:
@f1xpl thanks for the reply. I may have not put the point across correctly. What I was saying was that openauto need not support it directly, but having setup bluetooth HFP should automatically give me that. The existing head units in cars dont claim to support voip/app calls, but they work just because i have it paired to the car's bluetooth module as a call and media client. So when a whatsapp call comes in, the bluetooth device thinks there is a call coming in and shows call control UI. So today in openauto, what does setting the HPA profile on the rpi's external bluetooth module enable? Is it for the audio to be passed though the head unit?

f1xpl, 2018-07-14:
OpenAuto's bluetooth settings are related only to the Android Auto. OpenAuto itself does not use bluetooth. Functionality you mentioned should be a part of OFono - as far as I know OFono exposes some interface to control HFP/A2DP but you need to check it in the documentation.

closed 2018-07-17 by f1xpl