f1xpl/openauto #116: QXcbConnection: Could not connect to display

pprotschka, 2018-07-14:
compiled successfully and trying to launch openauto. I want to output to a regular screen via hmdi for now. (1080p)

upon launch im getting: **"QXcbConnection: Could not connect to display"** 
appreciate any help. i compiled when the RASPI  didnt have a desktop environment yet. Could that be the issue?

f1xpl, 2018-07-14:
It is not possible to run OpenAuto without desktop environment or Qt without EGL support.

pprotschka, 2018-07-15:
hey f1xpl, thanks for the reply.
of course i installed the desktop after (upgraded from headless) however this still didnt resolve the issue also reboots didnt fix the issue. I also ran under root instead of pi so perhaps things got messy there.

after starting form a band new image (RASPBIAN STRETCH WITH DESKTOP) i got the same error after i can this script... (novasprits kindly has made an auto install script that does the build automatically)
https://github.com/novaspirit/androidauto_rpi_install/blob/master/install.sh
i then rebooted and voila everything worked as excpected.

to summarize:
-> Do not start with headless and upgrade to desktop, start with desktop image right away.
-> install, build, run as User Pi not as root.

rdy for close.

closed 2018-07-15 by pprotschka