f1xpl/openauto #119: Failure running openauto on Google Pixel.

jonsmirl, 2018-07-25:
My first try running OpenAuto, I can't get it going. I receive these errors when AA loads on my Google Pixel. Using Linux as a host.

2018-07-25 17:38:55.207364] [0x00007fe6fedaa700] [info]    [OpenAuto] [App] Waiting for device...
[2018-07-25 17:38:55.209826] [0x00007fe6fe5a9700] [info]    [OpenAuto] [App] Devices enumeration result: true
[2018-07-25 17:38:55.874652] [0x00007fe6fedaa700] [info]    [OpenAuto] [App] Device connected.
[2018-07-25 17:38:55.884638] [0x00007fe70ea14840] [debug]   [OpenAuto] [AudioInput] create.
[2018-07-25 17:38:55.892837] [0x00007fe70ea14840] [debug]   [OpenAuto] [QtVideoOutput] create.
[2018-07-25 17:38:55.935615] [0x00007fe6fedaa700] [info]    [OpenAuto] [AndroidAutoEntity] start.
[2018-07-25 17:38:55.935632] [0x00007fe6fedaa700] [info]    [OpenAuto] [AudioInputService] start.
[2018-07-25 17:38:55.936324] [0x00007fe6fedaa700] [info]    [OpenAuto] [AudioService] start, channel: MEDIA_AUDIO
[2018-07-25 17:38:55.936432] [0x00007fe6fedaa700] [info]    [OpenAuto] [AudioService] start, channel: SPEECH_AUDIO
[2018-07-25 17:38:55.936476] [0x00007fe6fedaa700] [info]    [OpenAuto] [AudioService] start, channel: SYSTEM_AUDIO
[2018-07-25 17:38:55.936500] [0x00007fe6fedaa700] [info]    [OpenAuto] [SensorService] start.
[2018-07-25 17:38:55.936587] [0x00007fe6fedaa700] [info]    [OpenAuto] [VideoService] start.
[2018-07-25 17:38:55.936662] [0x00007fe6fedaa700] [info]    [OpenAuto] [BluetoothService] start.
[2018-07-25 17:38:55.936748] [0x00007fe6fedaa700] [info]    [OpenAuto] [InputService] start.
[2018-07-25 17:39:05.937655] [0x00007fe6fe5a9700] [error]   [OpenAuto] [AndroidAutoEntity] ping timer exceeded.
[2018-07-25 17:39:05.937707] [0x00007fe6fe5a9700] [info]    [OpenAuto] [App] quit.
[2018-07-25 17:39:05.937725] [0x00007fe6fe5a9700] [info]    [OpenAuto] [AndroidAutoEntity] stop.
[2018-07-25 17:39:05.937746] [0x00007fe6fe5a9700] [info]    [OpenAuto] [AudioInputService] stop.
[2018-07-25 17:39:05.937808] [0x00007fe6fe5a9700] [info]    [OpenAuto] [AudioService] stop, channel: MEDIA_AUDIO
[2018-07-25 17:39:05.937837] [0x00007fe6fe5a9700] [info]    [OpenAuto] [AudioService] stop, channel: SPEECH_AUDIO
[2018-07-25 17:39:05.937855] [0x00007fe6fe5a9700] [info]    [OpenAuto] [AudioService] stop, channel: SYSTEM_AUDIO
[2018-07-25 17:39:05.937875] [0x00007fe6fe5a9700] [info]    [OpenAuto] [SensorService] stop.
[2018-07-25 17:39:05.937900] [0x00007fe6fe5a9700] [info]    [OpenAuto] [VideoService] stop.
[2018-07-25 17:39:05.937933] [0x00007fe6fe5a9700] [info]    [OpenAuto] [BluetoothService] stop.
[2018-07-25 17:39:05.937960] [0x00007fe6fe5a9700] [info]    [OpenAuto] [InputService] stop.
[2018-07-25 17:39:05.937977] [0x00007fe6fe5a9700] [info]    [OpenAuto] [InputDevice] stop.
[2018-07-25 17:39:05.938524] [0x00007fe6fe5a9700] [info]    [OpenAuto] [App] Waiting for device...
[2018-07-25 17:39:05.938799] [0x00007fe6fe5a9700] [error]   [OpenAuto] [AndroidAutoEntity] channel error: AaSdk error code: 30, native code: 0
[2018-07-25 17:39:05.939033] [0x00007fe6fd5a7700] [error]   [OpenAuto] [AndroidAutoEntity] channel error: AaSdk error code: 30, native code: 0
[2018-07-25 17:39:05.939095] [0x00007fe6fd5a7700] [debug]   [OpenAuto] [AndroidAutoEntity] destroy.
[2018-07-25 17:39:05.939133] [0x00007fe6fe5a9700] [error]   [OpenAuto] [SensorService] channel error: AaSdk error code: 30, native code: 0
[2018-07-25 17:39:05.939167] [0x00007fe6fedaa700] [error]   [OpenAuto] [AudioService] channel error: AaSdk error code: 30, native code: 0, channel: MEDIA_AUDIO
[2018-07-25 17:39:05.939206] [0x00007fe6fd5a7700] [error]   [OpenAuto] [AudioInputService] channel error: AaSdk error code: 30, native code: 0
[2018-07-25 17:39:05.939232] [0x00007fe6fe5a9700] [error]   [OpenAuto] [AudioService] channel error: AaSdk error code: 30, native code: 0, channel: SPEECH_AUDIO
[2018-07-25 17:39:05.939260] [0x00007fe6fdda8700] [error]   [OpenAuto] [BluetoothService] channel error: AaSdk error code: 30, native code: 0
[2018-07-25 17:39:05.939344] [0x00007fe6fd5a7700] [error]   [OpenAuto] [AudioService] channel error: AaSdk error code: 30, native code: 0, channel: SYSTEM_AUDIO
[2018-07-25 17:39:05.939375] [0x00007fe6fedaa700] [error]   [OpenAuto] [VideoService] channel error: AaSdk error code: 30, native code: 0
[2018-07-25 17:39:05.939400] [0x00007fe6fe5a9700] [error]   [OpenAuto] [SensorService] channel error: AaSdk error code: 30, native code: 0
^C

jonsmirl, 2018-07-25:
Maybe this error is more useful....

[2018-07-25 17:48:35.537220] [0x00007f3ac37fe700] [info]    [OpenAuto] [InputDevice] start.
[2018-07-25 17:48:35.537248] [0x00007f3ac37fe700] [info]    [OpenAuto] [InputService] binding request, status: 0
[2018-07-25 17:48:35.539277] [0x00007f3ac37fe700] [info]    [OpenAuto] [SensorService] sensor start request, type: 10
[2018-07-25 17:48:35.550653] [0x00007f3ac8ae2700] [info]    [OpenAuto] [AudioService] open status: 0, channel: MEDIA_AUDIO
[2018-07-25 17:48:35.552499] [0x00007f3ac37fe700] [info]    [OpenAuto] [AudioService] setup request, channel: MEDIA_AUDIO, config index: 1
[2018-07-25 17:48:35.552556] [0x00007f3ac37fe700] [info]    [OpenAuto] [AudioService] setup status: 2, channel: MEDIA_AUDIO
[2018-07-25 17:48:35.554602] [0x00007f3ac92e3700] [info]    [OpenAuto] [AudioService] open status: 0, channel: SPEECH_AUDIO
[2018-07-25 17:48:35.555824] [0x00007f3ac3fff700] [info]    [OpenAuto] [AudioService] open status: 0, channel: SYSTEM_AUDIO
[2018-07-25 17:48:35.559772] [0x00007f3ac8ae2700] [info]    [OpenAuto] [AudioService] setup request, channel: SPEECH_AUDIO, config index: 1
[2018-07-25 17:48:35.559819] [0x00007f3ac8ae2700] [info]    [OpenAuto] [AudioService] setup status: 2, channel: SPEECH_AUDIO
[2018-07-25 17:48:35.562750] [0x00007f3ac92e3700] [info]    [OpenAuto] [AudioService] setup request, channel: SYSTEM_AUDIO, config index: 1
[2018-07-25 17:48:35.562787] [0x00007f3ac92e3700] [info]    [OpenAuto] [AudioService] setup status: 2, channel: SYSTEM_AUDIO
[2018-07-25 17:48:35.665669] [0x00007f3ac3fff700] [info]    [OpenAuto] [VideoService] start indication, session: 0
Error: "Internal data stream error."
appsrc: push buffer wrong state
appsrc: push buffer wrong state
appsrc: push buffer wrong state
appsrc: push buffer wrong state
appsrc: push buffer wrong state
appsrc: push buffer wrong state
appsrc: push buffer wrong state
appsrc: push buffer wrong state
appsrc: push buffer wrong state
appsrc: push buffer wrong state
appsrc: push buffer wrong state
appsrc: push buffer wrong state
appsrc: push buffer wrong state
appsrc: push buffer wrong state
appsrc: push buffer wrong state
appsrc: push buffer wrong state

closed 2019-01-17 by f1xpl

rafaelgil83, 2019-04-14:
Did you get any solution for this? I'm also having problems from rpi and Google Pixel