f1xpl/openauto #121: Google assistant popping noises

rosstadude, 2018-08-01:
I have everything running just fine in my car. I love this project!!! 

There's only one problem. When I use the wake word, the audio response pops and cracks. If I say: OK Google, what's the weather today- the answer comes back with a significant amount of popping noise and crackling. 

However, I can listen to podcasts, navigation, pandora, google Play music, etc without any audio issues. 

I am using a RPi 3, with a ugreen USB Soundcard connected to the Aux jack of my Pioneer head unit and a mic plugged into that same sound card. Even if I unplug the Mic, the popping crackling noises persist in Google assistant. I have followed the suggestion on the wiki page for pulse audio, and this has not resolved it. I am using the RPi 7" touchscreen and a mauseberry circuit power supply. 

Oh, and my phone is the original Pixel XL running Android 8.1.0 with the July 5, security patch. 

Does anyone have an insight to offer? Thank you!

rubenverhoef, 2018-08-20:
I have the same issue, also the Google maps navigation voice is giving popping noise. This voice is also out of sync when using for a while and sometimes stops in the middle of the sentence. And continues after a while. So after some time she is not saying the right things.

kovmarci86, 2018-09-01:
Hi,
For me, this helped:
https://github.com/opencardev/crankshaft/issues/208

rubenverhoef, 2018-09-02:
Switching to RT audio and that commit works for me also:)

closed 2019-01-17 by f1xpl