f1xpl/openauto #123: Question: android auto gui overlay

hawkeyexp, 2018-08-03:
Hi, is it generally possible to add a button/slider as overlay for the android auto gui if device is connected?
Would be really helpful to have a way to send a command to underlaying system - for example to create volume control commands etc.

Thx

f1xpl, 2018-08-03:
What is the desired platform? In case of regular linux (x86, x64) it can be easly achived via WM. In case of RPI it can be done via GPU driver.

hawkeyexp, 2018-08-04:
it's planed for rpi to add accessible volume controls to gui if android auto is projecting

f1xpl, 2018-09-11:
Overlay of Android Auto UI is supported in OpenAuto Pro (new project that I am involved in). There is an overview video at https://www.youtube.com/watch?v=9sTOMI1qTiA to see how it works.

closed 2018-09-13 by hawkeyexp