f1xpl/openauto #125: Phone fails to connect

sethraymond, 2018-08-11:
Plugging my phone into my Pi3 does not move past the landing page. Here is the given error:

```
libEGL warning: DRI2: failed to authenticate
[2018-08-09 07:01:07.745348] [0x723d2000] [warning] [OpenAuto] [Configuration] failed to read configuration file: openauto.ini, error: openauto.ini: cannot open file. Using default configuration.
[2018-08-09 07:01:07.795677] [0x723d2000] [warning] [OpenAuto] [RecentAddressesList] failed to read configuration file: openauto_wifi_recent.ini, error: openauto_wifi_recent.ini: cannot open file. Empty list will be used.
[2018-08-09 07:01:07.825249] [0x6f9ff310] [info]    [OpenAuto] [App] Waiting for device...
[2018-08-09 07:01:07.825784] [0x6f9ff310] [info]    [OpenAuto] [App] Devices enumeration result: false
```

f1xpl, 2018-08-11:
Looks like wrong configuration.

sethraymond, 2018-08-11:
I've run it multiple times with other configurations and it gives the same behavior... the problem is that the phone isn't being recognized.

sethraymond, 2018-08-11:
I have taken the steps listed here: https://ubuntuforums.org/showthread.php?t=2226702

As a note, the rules to download from https://github.com/f1xpl/openauto/issues/59 don't seem to be sufficient (at least, it wasn't for a Google Pixel XL, I had to add my own rule).

However, now the error is changing. When I plug my phone in, I get a popup error saying "No MTP devices found" and `autoapp` crashes. Here is the new log:

```
[2018-08-10 23:47:02.927051] [0x7248e000] [warning] [OpenAuto] [RecentAddressesList] failed to read configuration file: openauto_wifi_recent.ini, error: openauto_wifi_recent.ini: cannot open file. Empty list will be used.
[2018-08-10 23:47:02.951817] [0x6faff310] [info]    [OpenAuto] [App] Waiting for device...
[2018-08-10 23:47:02.956080] [0x6faff310] [info]    [OpenAuto] [App] Devices enumeration result: true
[2018-08-10 23:47:09.284579] [0x6faff310] [info]    [OpenAuto] [App] Device connected.
[2018-08-10 23:47:09.308482] [0x7248e000] [debug]   [OpenAuto] [AudioInput] create.
[2018-08-10 23:47:09.584386] [0x6faff310] [info]    [OpenAuto] [AndroidAutoEntity] start.
[2018-08-10 23:47:09.584439] [0x6faff310] [info]    [OpenAuto] [AudioInputService] start.
[2018-08-10 23:47:09.584798] [0x6faff310] [info]    [OpenAuto] [AudioService] start, channel: MEDIA_AUDIO
[2018-08-10 23:47:09.584858] [0x6faff310] [info]    [OpenAuto] [AudioService] start, channel: SPEECH_AUDIO
[2018-08-10 23:47:09.584888] [0x6faff310] [info]    [OpenAuto] [AudioService] start, channel: SYSTEM_AUDIO
[2018-08-10 23:47:09.584919] [0x6faff310] [info]    [OpenAuto] [SensorService] start.
[2018-08-10 23:47:09.584957] [0x6faff310] [info]    [OpenAuto] [VideoService] start.
[2018-08-10 23:47:09.584996] [0x6faff310] [info]    [OpenAuto] [BluetoothService] start.
[2018-08-10 23:47:09.585037] [0x6faff310] [info]    [OpenAuto] [InputService] start.
[2018-08-10 23:47:09.587041] [0x6f2ff310] [info]    [OpenAuto] [AndroidAutoEntity] version response, version: 1.5, status: 0
[2018-08-10 23:47:09.587095] [0x6f2ff310] [info]    [OpenAuto] [AndroidAutoEntity] Begin handshake.
[2018-08-10 23:47:09.602081] [0x6f2ff310] [info]    [OpenAuto] [AndroidAutoEntity] Handshake, size: 2290
[2018-08-10 23:47:09.641565] [0x6f2ff310] [info]    [OpenAuto] [AndroidAutoEntity] continue handshake.
[2018-08-10 23:47:09.651738] [0x6faff310] [info]    [OpenAuto] [AndroidAutoEntity] Handshake, size: 51
[2018-08-10 23:47:09.652177] [0x6faff310] [info]    [OpenAuto] [AndroidAutoEntity] Auth completed.
[2018-08-10 23:47:09.686065] [0x6e2ff310] [info]    [OpenAuto] [AndroidAutoEntity] Discovery request, device name: Android, brand: Google Pixel XL
[2018-08-10 23:47:09.686132] [0x6e2ff310] [info]    [OpenAuto] [AudioInputService] fill features.
[2018-08-10 23:47:09.686206] [0x6e2ff310] [info]    [OpenAuto] [AudioService] fill features, channel: MEDIA_AUDIO
[2018-08-10 23:47:09.686237] [0x6e2ff310] [info]    [OpenAuto] [AudioService] fill features, channel: SPEECH_AUDIO
[2018-08-10 23:47:09.686261] [0x6e2ff310] [info]    [OpenAuto] [AudioService] fill features, channel: SYSTEM_AUDIO
[2018-08-10 23:47:09.686282] [0x6e2ff310] [info]    [OpenAuto] [SensorService] fill features.
[2018-08-10 23:47:09.686318] [0x6e2ff310] [info]    [OpenAuto] [VideoService] fill features.
[2018-08-10 23:47:09.686348] [0x6e2ff310] [info]    [OpenAuto] [BluetoothService] fill features
[2018-08-10 23:47:09.686368] [0x6e2ff310] [info]    [OpenAuto] [InputService] fill features.
[2018-08-10 23:47:09.795749] [0x6f2ff310] [info]    [OpenAuto] [AndroidAutoEntity] requested audio focus, type: 4
[2018-08-10 23:47:09.795801] [0x6f2ff310] [info]    [OpenAuto] [AndroidAutoEntity] audio focus state: 3
[2018-08-10 23:47:09.799963] [0x6f2ff310] [info]    [OpenAuto] [AudioInputService] open request, priority: 0
[2018-08-10 23:47:09.799992] [0x6f2ff310] [info]    [OpenAuto] [AudioInputService] open status: 0
[2018-08-10 23:47:09.800484] [0x6f2ff310] [info]    [OpenAuto] [AudioService] open request, channel: MEDIA_AUDIO, priority: 0
[2018-08-10 23:47:09.800518] [0x6f2ff310] [debug]   [OpenAuto] [AudioService] channel: MEDIA_AUDIO audio output sample rate: 48000, sample size: 16, channel count: 2
[2018-08-10 23:47:09.800744] [0x6e2ff310] [info]    [OpenAuto] [AudioService] open request, channel: SPEECH_AUDIO, priority: 0
[2018-08-10 23:47:09.800776] [0x6e2ff310] [debug]   [OpenAuto] [AudioService] channel: SPEECH_AUDIO audio output sample rate: 16000, sample size: 16, channel count: 1
[2018-08-10 23:47:09.801094] [0x6eaff310] [info]    [OpenAuto] [AudioService] open request, channel: SYSTEM_AUDIO, priority: 0
[2018-08-10 23:47:09.801276] [0x6eaff310] [debug]   [OpenAuto] [AudioService] channel: SYSTEM_AUDIO audio output sample rate: 16000, sample size: 16, channel count: 1
[2018-08-10 23:47:09.801835] [0x6faff310] [info]    [OpenAuto] [SensorService] open request, priority: 0
[2018-08-10 23:47:09.801865] [0x6faff310] [info]    [OpenAuto] [SensorService] open status: 0
[2018-08-10 23:47:09.802387] [0x6faff310] [info]    [OpenAuto] [VideoService] open request, priority: 0
[2018-08-10 23:47:09.802412] [0x6faff310] [info]    [OpenAuto] [OMXVideoOutput] open.
* failed to open vchiq instance
```

sethraymond, 2018-08-11:
The information post above is immediately followed by

```
PulseAudioService: pa_context_connect() failed
QObject::~QObject: Timers cannot be stopped from another thread
```

f1xpl, 2018-08-11:
Please follow official build and setup instructions.

sethraymond, 2018-08-11:
I have followed your official build instructions. I've now disabled the [HFP and A2DP setup for Raspberry Pi 3](https://github.com/f1xpl/openauto/wiki/HFP-and-A2DP-setup-for-Raspberry-PI-3) and my error is now:

```
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injectory<boost::log::v2_mt_posix::system_error> >'
    what():  Failed to set TLS value: Invalid argument
terminate called recursively
```

closed 2019-01-17 by f1xpl