f1xpl/openauto #126: No audio with pulseaudio on 3,5 mm jack

kom94, 2018-08-11:
I have installed open auto on raspbian using a raspberry pi 3 B with the official 7" touchscreen.

I am trying to get audio out of the raspberry pi by a 3,5 mm jack on the raspberry pi, but i won't get any audio from open auto.
My audio works on Kodi using analog audio but when i switch to ALSA: .. pulseaudio i wont get any audio,
with a quick search i found out that open auto use pulseaudio so i tried to install it with success but still no audio by Open auto or Kodi, i have search for more try different things i bought an usb sound card but still no success.

can someone help me out, do i need an audio hat to get pulseaudio to work??

i want to install my raspberrypi in my car with a 2.1 system, i was thinking to connect the pi to the aux input on my radio.

[image]
[image]
[image]

mentalinc, 2018-08-11:
Have you followed the two wiki posts?
You need a USB sound card...
https://github.com/f1xpl/openauto/wiki/Raspberry-PI-3-requirements

https://github.com/f1xpl/openauto/wiki/Pulseaudio-configuration

kom94, 2018-08-12:
Well for some weird reason it works now!
i maked a file: "/etc/asound.conf" put in this code:

`pi@raspberrypi ~ $ cat ~/.asoundrc
pcm.!default {
        type hw
        card 0
        device 0
}
ctl.!default {
        type hw
        card 0
        device 0
}`

Got an error from kodi when i reboot the raspberry pi, so i removed the file en reboot again and it works it's just weird.
It works on kodi and open auto but not on the start video on raspbian but that's no big deal -_-

MalteHillmann, 2018-08-13:
I have a similar problem:
I tested around with OpenAuto and the sound output works from the internal rpi3 3,5" jack. 
Now i bought an external USB-Soundcard. It works nice in VLC. But in OpenAuto it does not work. OpenAuto plays the sound always on the internal 3,5" jack of the rpi3.

MalteHillmann, 2018-08-13:
I managed to set the output device by the name of the sound card.
Try this: https://wiki.archlinux.org/index.php/PulseAudio/Examples#Set_the_default_output_source

closed 2019-01-17 by f1xpl