f1xpl/openauto #129: Keyboard Only Usage

black-energie, 2018-08-16:
I cannot fully interact with AA using only the keyboard (I do not plan on using a touch-screen).  I can go through most menus using the wheel-left and wheel-right keys, but have issues when it comes to the navigation screen.  In this case wheel-left and wheel-right keys can only be used for zoom, so I'm unable to pan the screen, setup / cancel routes, etc.  Additionally there doesn't appear to be a way to interact with the on-screen keyboard.

I saw the video on the f1x.studio youtube page "Android Auto integration with Audi MMI 2G" where a rotary knob was used to successfully interact with a modified navigation interface.  Is the modified interface something that will be coming to this project?  Also will entry via the on-screen keyboard be supported?  

Additionally I think it would also be very useful add more keyboard bindings with a customizable mapping.  Particularly to support the full qwerty layout for text entry (i.e. to input the destination), and to map other keys to remaining functions (i.e. track play/pause/skip).

CluelessGenius, 2018-12-20:
im also not gonna use a touch screen but im still in the planning phase and didnt even test this yet. but that sounds bad.

marcimix, 2019-05-14:
I'm wanna buy the rpi so I have doubt about keyboard via bluetooth? How it's works? 
Yeah we need somehow mapping  the buttons easy.