f1xpl/openauto #132: Microphone: only "ok google" works. Everything said after the Codeword isn't recognized.

gillohner, 2018-08-20:
I connected a USB-Microphone to my raspberry Pi. It works in raspbian(I can record with it). When I use OpenAuto and say "ok google" google now pops up, but the 3 dots just stand still as if I don't say anything. As soon as I stop saying something google now closes. Did anyone had this problem before? Can somebody please help me?

Thanks

swavans, 2018-08-20:
I had similair issues, and noticed that most of the times the ok google detection was done on my phone for some reason. 

A couple of things to check:
- sound quality of the microphone (how clearly can you here your own voice and how much distortion is there)
- is the microphone the default device?

you can also try to wait a couple of seconds before giving the actual command, sometimes it's a bit laggy.

I believe there is also a wiki page on this

WarDrake, 2018-08-20:
#68

closed 2018-08-23 by f1xpl