f1xpl/openauto #137: OpenAuto Pro Android Auto (AA) interface

carcery, 2018-09-14:
I noticed the following when I launch AA...

the AA interface shows up, but I am unable to tap on any of the menu items (maps, music, etc)... however I can still see the very top of the openauto screen behind AA.

on the top left side I see the name of the last openauto menu I had before launching AA. when I try to tap items on the AA  screen it seems I am actually selecting items from the openauto screen behind AA, I think that is the case because when I touch a spot on the right side of the screen I can see the top left change name to "applications", "settings", "help", etc.

on the top right side I see the arrow head that allows me to pull down the audio volume sliding bar, if I touch the arrow head I see that the menu expands down but falls behind AA interface and I am unable to see the sliding bars...

it seems the problem is not with the touchscreen functionality but rather with being able to interact with AA once it launches and projects to the RPI.
I really hope I am explaining this correctly

My setup is a RPi3 B with the official 7" touchscreen running OpenAuto Pro

carcery, 2018-09-14:
Nexus 6P with android 8.1.0 stock and not rooted.
Android Auto  3.5.853024-release

attached is my log from /home/pi/.cache/lxsession/LXDE-pi/

I connected the 
[run2.log](https://github.com/f1xpl/openauto/files/2384831/run2.log)
phone to Pi at 13:12 disconnected 20 secs later... then connected again at 13:13 then disconnected 15 seconds later.

f1xpl, 2018-09-14:
Hello @chrisfromwa, In case of OpenAuto Pro issues please contact us via e-mail. It would be easier to track the issue. Also mixing issues of Pro and Open Source version may be misleading for the users. Thank you.

closed 2018-09-14 by f1xpl