f1xpl/openauto #140: Can't get my phone to conecct

Theklinto, 2018-09-23:
I've just installed openauto on my Raspberry pi 3 B+ with the official 7" touchscreen.
I can't get my phone to connect.
I have followed the build instructions for assdk and openauto and have gotten no errors in the process.
I start openauto on the pi, and connect my phone via USB. The AndroidAuto opens on my phone and i can see the service start, but no response on the pi.

Pi is running latest Raspbian with all updates and upgrades.
My phone is a Huawei P20 Lite running android 8.0 with AndroidAuto v3.4.5

I don't have any other phones to test with.

Bobet21, 2018-10-02:
You can try to connect over wifi seems to have worked for other people - sometimes a reinstall of openauto has worked as well - for me none of those solutions worked 100% (in wifi my phone detects openauto but does not stream the video)

Bobet21, 2018-11-23:
Also make sur your phone is ALSO connected in bluetooth with your RPI -it worked for me

jonfreva, 2018-12-03:
Maybe you didn't create the udev rules, they aren't mentioned by the build instructions explicitly. You can find more info in the wiki [https://github.com/f1xpl/openauto/wiki/udev-rules-(USB-permissions)]

Pobyeda, 2019-01-05:
try to start open auto like this
sudo su
cd ~
cd home/pi/openauto/bin
./autoapp

ugifractal, 2019-01-05:
@Pobyeda few weeks ago, I also got problem opening autoapp, the text on GUI seem missing and unable to connect the phone. Then I tried using sudo command. and working great.

Theklinto, 2019-01-07:
I have fixed the issue. Don't know what. But installing the second newest release helped me. Now I can use over USB and WiFi