f1xpl/openauto #142: touch screen doesn't work at first time

computeridea, 2018-09-25:
hi, i have a custom touchscreen, it work fine evrywhere, in linux or in the waiting screen, but when i connect the phone, open auto start correctly, but open out don't send information about the touch to the phone, if i try to plugin and plug out 2/3 times, it start to work.. 
i reapet, i can use always the touch in open out or raspian, it's just as the touch information donesn't reacht the phone .
i tried with 2 phone, lg g4 and samsung a2 2017, same issues.. 
any advice please?

ppatacchiola, 2018-10-23:
Same here, tried changing DPI and resolution in OpenAuto settings buttare doesn't fix the issue.
My setup: LCD waveshare 7" 
Oneplus 5 with Stock Android 9

f1xpl, 2019-01-17:
Not reproducible.

closed 2019-01-17 by f1xpl