f1xpl/openauto #145: Phone doesn't connect

ppatacchiola, 2018-11-02:
It's been a week since my phone cant connect anymore to my raspberry, when i plug the usb into my phone i get this screen:  https://drive.google.com/file/d/1EblCV7s_y-VSyafPKQ9lB8d2vv76lQUP/view?usp=drivesdk

(Translation: comunication erorr n2)

Tried to reinstalling Android Auto app in my phone and resetting to default settings on OpenAuto settings page but that doesn't help.. 

Anyone had this issue before?

Bobet21, 2018-11-17:
Same issue...I have upgraded my Androidauto version and since then i cannot connect anymmore

ppatacchiola, 2018-11-17:
@bobet21 I fixed the problem by checking the option "allow unkown sources" in the developper settings of Android Auto app

Bobet21, 2018-11-23:
Thanks, for me i needed to Connect to Bluetooth in addition to the USB cable

Bishamon1987, 2019-01-01:
I just got the latest version of Android Auto from apkmirror.com, since its not officialy out in the Netherlands, but it doesnt seem to work.
So based on which version is this all based?

abchet, 2019-01-10:
@ppatacchiola i fixed mine by https://github.com/f1xpl/openauto/wiki/udev-rules-(USB-permissions)
I don't know why but it fixed my issue

closed 2019-01-17 by f1xpl