f1xpl/openauto #149: Additional button bindings

chucktastic, 2018-12-06:
Looking at the supported keyboard button bindings here:

https://github.com/f1xpl/openauto/wiki/Keyboard-button-bindings

Is it possible to add a couple additional bindings?  It would be nice to have bindings for navigation and media.

CluelessGenius, 2018-12-20:
thinking about the same thing. actually spent some time now looking into this. but cant find if theres android auto specific keycodes somewhere. only android general keycodes which i would have to test through all of them to see what android auto reacts to. and if none of them do anything alse besides the already implemented. then i guess android auto doesnt support it. im also thinking if mouse events could be simulated at certain coordinates so at least i could map those functions to key bindings myself.

CluelessGenius, 2018-12-20:
so from the looks of it if im understanding this right qt simply translates touch input to mouse events anyway and thats why openauto converts qt mouse input events to touch events to send to android auto.
if that is true theoretically i assume i should be able to just emulate mouse events on the itself in lets say a simple python script and be able to tap anywhere on the screen programmatically / i.e. with keyboard bindings. for the bottom row of menu item this should be simple since they never move but for more complex interaction like lets say.. pausing music while in navigation screen or ending a call while not in phone screen id still have to rely on keyboard shortcuts and interacting with the sctive screen itself would require me to keep track of where the focus/cursor/mouse is at any point in time plus which screen is active. could turn out pretty tricky

CluelessGenius, 2018-12-21:
tested and works. even though the mouse cursor isnt visible during open auto it is however functional which makes it easy to trigger mouse events on certain positions and the visibility thing actually kinda work in my favour that way

chucktastic, 2018-12-21:
That's pretty cool.  How does that work if the touchscreen is not enabled in the input settings?  I notice that when the touchscreen is not enabled in the input tab that AA does not respond to touches on my development RPI (which has an official RPI screen).  My CS install in the car does not have a touchscreen at all and I rely on a rotary encoder and buttons through a keyboard emulator board.  This setup works much better in my particular car since the screen is well out of reach to touch.

CluelessGenius, 2018-12-21:
well you enable it. in settings.

chucktastic, 2018-12-23:
Would you be open to sharing your script to trigger mouse events from keypresses?  This seems like this would work for pretty well for my use case.