f1xpl/openauto #152: Stuck at "Waiting for Device "

OryanBaker, 2018-12-29:
Fresh setup on RPi3+ Raspbian Stretch Full using HDMI screen, wireless keyboard & mouse while doing full setup. Program loads up no problems, connecting phone (ZTE Axon 7) through wifi or usb does not change entry screen.

Attached is a log on running:

> [2018-12-28 23:46:01.281202] [0x6faff310] [info]    [OpenAuto] [App] Waiting for device...
[2018-12-28 23:46:01.284063] [0x6e2ff310] [info]    [OpenAuto] [App] Devices enumeration result: false
[2018-12-28 23:46:19.391990] [0x6e2ff310] [info]    [OpenAuto] [App] Device connected.
[2018-12-28 23:46:19.422045] [0x72428000] [debug]   [OpenAuto] [AudioInput] create.
[2018-12-28 23:46:19.437300] [0x72428000] [debug]   [OpenAuto] [LocalBluetoothDevice] create.
[2018-12-28 23:46:19.477861] [0x6e2ff310] [info]    [OpenAuto] [AndroidAutoEntity] start.
[2018-12-28 23:46:19.477942] [0x6e2ff310] [info]    [OpenAuto] [AudioInputService] start.
[2018-12-28 23:46:19.478530] [0x6e2ff310] [info]    [OpenAuto] [AudioService] start, channel: MEDIA_AUDIO
[2018-12-28 23:46:19.478613] [0x6e2ff310] [info]    [OpenAuto] [AudioService] start, channel: SPEECH_AUDIO
[2018-12-28 23:46:19.478668] [0x6e2ff310] [info]    [OpenAuto] [AudioService] start, channel: SYSTEM_AUDIO
[2018-12-28 23:46:19.478723] [0x6e2ff310] [info]    [OpenAuto] [SensorService] start.
[2018-12-28 23:46:19.478788] [0x6e2ff310] [info]    [OpenAuto] [VideoService] start.
[2018-12-28 23:46:19.478847] [0x6e2ff310] [info]    [OpenAuto] [BluetoothService] start.
[2018-12-28 23:46:19.478928] [0x6e2ff310] [info]    [OpenAuto] [InputService] start.
[2018-12-28 23:46:29.480440] [0x6f2ff310] [error]   [OpenAuto] [AndroidAutoEntity] ping timer exceeded.
[2018-12-28 23:46:29.480589] [0x6f2ff310] [info]    [OpenAuto] [App] quit.
[2018-12-28 23:46:29.480664] [0x6f2ff310] [info]    [OpenAuto] [AndroidAutoEntity] stop.
[2018-12-28 23:46:29.480729] [0x6f2ff310] [info]    [OpenAuto] [AudioInputService] stop.
[2018-12-28 23:46:29.480935] [0x6f2ff310] [info]    [OpenAuto] [AudioService] stop, channel: MEDIA_AUDIO
[2018-12-28 23:46:29.480995] [0x6f2ff310] [info]    [OpenAuto] [AudioService] stop, channel: SPEECH_AUDIO
[2018-12-28 23:46:29.481045] [0x6f2ff310] [info]    [OpenAuto] [AudioService] stop, channel: SYSTEM_AUDIO
[2018-12-28 23:46:29.481109] [0x6f2ff310] [info]    [OpenAuto] [SensorService] stop.
[2018-12-28 23:46:29.481170] [0x6f2ff310] [info]    [OpenAuto] [VideoService] stop.
[2018-12-28 23:46:29.481218] [0x6f2ff310] [info]    [OpenAuto] [OMXVideoOutput] stop.
[2018-12-28 23:46:29.481275] [0x6f2ff310] [info]    [OpenAuto] [BluetoothService] stop.
[2018-12-28 23:46:29.481339] [0x6f2ff310] [info]    [OpenAuto] [InputService] stop.
[2018-12-28 23:46:29.481386] [0x6f2ff310] [info]    [OpenAuto] [InputDevice] stop.
[2018-12-28 23:46:29.483038] [0x6f2ff310] [info]    [OpenAuto] [App] Waiting for device...
[2018-12-28 23:46:29.483154] [0x6f2ff310] [error]   [OpenAuto] [AndroidAutoEntity] channel error: AaSdk error code: 10, native code: 2
[2018-12-28 23:46:29.483212] [0x6f2ff310] [error]   [OpenAuto] [AndroidAutoEntity] channel error: AaSdk error code: 10, native code: 2
[2018-12-28 23:46:29.484075] [0x6f2ff310] [error]   [OpenAuto] [AndroidAutoEntity] channel error: AaSdk error code: 30, native code: 0
[2018-12-28 23:46:29.484151] [0x6f2ff310] [debug]   [OpenAuto] [AndroidAutoEntity] destroy.
[2018-12-28 23:46:29.484221] [0x6f2ff310] [error]   [OpenAuto] [SensorService] channel error: AaSdk error code: 30, native code: 0
[2018-12-28 23:46:29.484353] [0x6f2ff310] [error]   [OpenAuto] [BluetoothService] channel error: AaSdk error code: 30, native code: 0
[2018-12-28 23:46:29.484508] [0x6f2ff310] [error]   [OpenAuto] [VideoService] channel error: AaSdk error code: 30, native code: 0
[2018-12-28 23:46:29.484599] [0x6f2ff310] [error]   [OpenAuto] [SensorService] channel error: AaSdk error code: 30, native code: 0
[2018-12-28 23:46:29.484676] [0x6f2ff310] [error]   [OpenAuto] [AudioInputService] channel error: AaSdk error code: 30, native code: 0
[2018-12-28 23:46:29.484797] [0x6f2ff310] [error]   [OpenAuto] [AudioService] channel error: AaSdk error code: 30, native code: 0, channel: MEDIA_AUDIO
[2018-12-28 23:46:29.485203] [0x6f2ff310] [error]   [OpenAuto] [AudioService] channel error: AaSdk error code: 30, native code: 0, channel: SPEECH_AUDIO
[2018-12-28 23:46:29.485402] [0x6f2ff310] [error]   [OpenAuto] [AudioService] channel error: AaSdk error code: 30, native code: 0, channel: SYSTEM_AUDIO

OryanBaker, 2018-12-30:
Finally made headway today after work (automotive mechanic), I've now gotten down to only one issue, but I'm at a loss for why. Using wifi connection, as the USB C draws too much power I think, I only have the following code remaining:

> [2018-12-29 20:36:47.444245] [0x6fb98310] [info]    [OpenAuto] [App] Waiting for device...
[2018-12-29 20:36:47.459856] [0x6f398310] [info]    [OpenAuto] [App] Devices enumeration result: false
[2018-12-29 20:36:56.150574] [0x7259e000] [debug]   [OpenAuto] [AudioInput] create.
[2018-12-29 20:36:56.156551] [0x7259e000] [debug]   [OpenAuto] [QtAudioOutput] create.
[2018-12-29 20:36:56.158908] [0x7259e000] [debug]   [OpenAuto] [QtAudioOutput] create.
[2018-12-29 20:36:56.159290] [0x7259e000] [debug]   [OpenAuto] [QtAudioOutput] create.
[2018-12-29 20:36:56.159834] [0x7259e000] [debug]   [OpenAuto] [LocalBluetoothDevice] create.
[2018-12-29 20:36:56.203706] [0x6fb98310] [info]    [OpenAuto] [AndroidAutoEntity] start.
[2018-12-29 20:36:56.203791] [0x6fb98310] [info]    [OpenAuto] [AudioInputService] start.
[2018-12-29 20:36:56.204359] [0x6fb98310] [info]    [OpenAuto] [AudioService] start, channel: MEDIA_AUDIO
[2018-12-29 20:36:56.204440] [0x6fb98310] [info]    [OpenAuto] [AudioService] start, channel: SPEECH_AUDIO
[2018-12-29 20:36:56.204498] [0x6fb98310] [info]    [OpenAuto] [AudioService] start, channel: SYSTEM_AUDIO
[2018-12-29 20:36:56.204572] [0x6fb98310] [info]    [OpenAuto] [SensorService] start.
[2018-12-29 20:36:56.204636] [0x6fb98310] [info]    [OpenAuto] [VideoService] start.
[2018-12-29 20:36:56.204701] [0x6fb98310] [info]    [OpenAuto] [BluetoothService] start.
[2018-12-29 20:36:56.204767] [0x6fb98310] [info]    [OpenAuto] [InputService] start.
[2018-12-29 20:36:56.205621] [0x6f398310] [error]   [OpenAuto] [App] usb hub error: AaSdk error code: 30, native code: 0

OryanBaker, 2018-12-30:
> [2018-12-29 20:52:59.570012] [0x6fbff310] [info]    [OpenAuto] [App] Waiting for device...
[2018-12-29 20:52:59.573990] [0x6fbff310] [info]    [OpenAuto] [App] Devices enumeration result: false
[2018-12-29 20:53:15.964582] [0x7257b000] [debug]   [OpenAuto] [AudioInput] create.
[2018-12-29 20:53:15.977641] [0x7257b000] [debug]   [OpenAuto] [QtAudioOutput] create.
[2018-12-29 20:53:15.978871] [0x7257b000] [debug]   [OpenAuto] [QtAudioOutput] create.
[2018-12-29 20:53:15.979276] [0x7257b000] [debug]   [OpenAuto] [QtAudioOutput] create.
[2018-12-29 20:53:15.979880] [0x7257b000] [debug]   [OpenAuto] [LocalBluetoothDevice] create.
[2018-12-29 20:53:15.999953] [0x6ebff310] [info]    [OpenAuto] [AndroidAutoEntity] start.
[2018-12-29 20:53:15.999997] [0x6ebff310] [info]    [OpenAuto] [AudioInputService] start.
[2018-12-29 20:53:16.000300] [0x6ebff310] [info]    [OpenAuto] [AudioService] start, channel: MEDIA_AUDIO
[2018-12-29 20:53:16.000340] [0x6ebff310] [info]    [OpenAuto] [AudioService] start, channel: SPEECH_AUDIO
[2018-12-29 20:53:16.000365] [0x6ebff310] [info]    [OpenAuto] [AudioService] start, channel: SYSTEM_AUDIO
[2018-12-29 20:53:16.000399] [0x6ebff310] [info]    [OpenAuto] [SensorService] start.
[2018-12-29 20:53:16.000432] [0x6ebff310] [info]    [OpenAuto] [VideoService] start.
[2018-12-29 20:53:16.000464] [0x6ebff310] [info]    [OpenAuto] [BluetoothService] start.
[2018-12-29 20:53:16.000497] [0x6ebff310] [info]    [OpenAuto] [InputService] start.
[2018-12-29 20:53:16.000959] [0x6fbff310] [error]   [OpenAuto] [App] usb hub error: AaSdk error code: 30, native code: 0
[2018-12-29 20:53:16.086113] [0x6e3ff310] [info]    [OpenAuto] [AndroidAutoEntity] version response, version: 1.5, status: 0
[2018-12-29 20:53:16.086157] [0x6e3ff310] [info]    [OpenAuto] [AndroidAutoEntity] Begin handshake.
[2018-12-29 20:53:16.103746] [0x6fbff310] [info]    [OpenAuto] [AndroidAutoEntity] Handshake, size: 2284
[2018-12-29 20:53:16.143797] [0x6fbff310] [info]    [OpenAuto] [AndroidAutoEntity] continue handshake.
[2018-12-29 20:53:16.164314] [0x6fbff310] [info]    [OpenAuto] [AndroidAutoEntity] Handshake, size: 51
[2018-12-29 20:53:16.164825] [0x6fbff310] [info]    [OpenAuto] [AndroidAutoEntity] Auth completed.
[2018-12-29 20:53:16.174833] [0x6fbff310] [info]    [OpenAuto] [AndroidAutoEntity] Discovery request, device name: Android, brand: ZTE Axon 7
[2018-12-29 20:53:16.174911] [0x6fbff310] [info]    [OpenAuto] [AudioInputService] fill features.
[2018-12-29 20:53:16.174990] [0x6fbff310] [info]    [OpenAuto] [AudioService] fill features, channel: MEDIA_AUDIO
[2018-12-29 20:53:16.175022] [0x6fbff310] [info]    [OpenAuto] [AudioService] fill features, channel: SPEECH_AUDIO
[2018-12-29 20:53:16.175043] [0x6fbff310] [info]    [OpenAuto] [AudioService] fill features, channel: SYSTEM_AUDIO
[2018-12-29 20:53:16.175065] [0x6fbff310] [info]    [OpenAuto] [SensorService] fill features.
[2018-12-29 20:53:16.175102] [0x6fbff310] [info]    [OpenAuto] [VideoService] fill features.
[2018-12-29 20:53:16.175133] [0x6fbff310] [info]    [OpenAuto] [BluetoothService] fill features
[2018-12-29 20:53:16.176828] [0x6fbff310] [info]    [OpenAuto] [BluetoothService] sending local adapter adress: B8:27:EB:AA:09:82
[2018-12-29 20:53:16.178019] [0x6fbff310] [info]    [OpenAuto] [InputService] fill features.
[2018-12-29 20:53:16.311049] [0x6fbff310] [info]    [OpenAuto] [AndroidAutoEntity] requested audio focus, type: 4
[2018-12-29 20:53:16.311096] [0x6fbff310] [info]    [OpenAuto] [AndroidAutoEntity] audio focus state: 3
[2018-12-29 20:53:16.315667] [0x6fbff310] [info]    [OpenAuto] [AudioInputService] open request, priority: 0
[2018-12-29 20:53:16.315718] [0x6fbff310] [info]    [OpenAuto] [AudioInputService] open status: 0
[2018-12-29 20:53:16.315728] [0x6f3ff310] [info]    [OpenAuto] [AudioService] open request, channel: MEDIA_AUDIO, priority: 0
[2018-12-29 20:53:16.315778] [0x6f3ff310] [debug]   [OpenAuto] [AudioService] channel: MEDIA_AUDIO audio output sample rate: 48000, sample size: 16, channel count: 2
[2018-12-29 20:53:16.315838] [0x6f3ff310] [info]    [OpenAuto] [AudioService] open status: 0, channel: MEDIA_AUDIO
[2018-12-29 20:53:16.316229] [0x6fbff310] [info]    [OpenAuto] [AudioService] open request, channel: SPEECH_AUDIO, priority: 0
[2018-12-29 20:53:16.316307] [0x6fbff310] [debug]   [OpenAuto] [AudioService] channel: SPEECH_AUDIO audio output sample rate: 16000, sample size: 16, channel count: 1
[2018-12-29 20:53:16.316343] [0x6fbff310] [info]    [OpenAuto] [AudioService] open status: 0, channel: SPEECH_AUDIO
[2018-12-29 20:53:16.320717] [0x6fbff310] [info]    [OpenAuto] [AudioService] open request, channel: SYSTEM_AUDIO, priority: 0
[2018-12-29 20:53:16.320778] [0x6fbff310] [debug]   [OpenAuto] [AudioService] channel: SYSTEM_AUDIO audio output sample rate: 16000, sample size: 16, channel count: 1
[2018-12-29 20:53:16.320811] [0x6fbff310] [info]    [OpenAuto] [AudioService] open status: 0, channel: SYSTEM_AUDIO
[2018-12-29 20:53:16.324824] [0x6fbff310] [info]    [OpenAuto] [VideoService] open request, priority: 0
[2018-12-29 20:53:16.324938] [0x6e3ff310] [info]    [OpenAuto] [SensorService] open request, priority: 0
[2018-12-29 20:53:16.324971] [0x6e3ff310] [info]    [OpenAuto] [SensorService] open status: 0
[2018-12-29 20:53:16.324983] [0x6ebff310] [info]    [OpenAuto] [BluetoothService] open request, priority: 0
[2018-12-29 20:53:16.325016] [0x6ebff310] [info]    [OpenAuto] [BluetoothService] open status: 0
[... cut ...]
[2018-12-29 20:53:38.038130] [0x6ebff310] [error]   [OpenAuto] [AudioService] channel error: AaSdk error code: 30, native code: 0, channel: MEDIA_AUDIO
[2018-12-29 20:53:38.038164] [0x6fbff310] [error]   [OpenAuto] [BluetoothService] channel error: AaSdk error code: 30, native code: 0
[2018-12-29 20:53:38.038231] [0x6ebff310] [error]   [OpenAuto] [AudioInputService] channel error: AaSdk error code: 30, native code: 0
[2018-12-29 20:53:38.039330] [0x6f3ff310] [info]    [OpenAuto] [App] Waiting for device...
[2018-12-29 20:53:38.039452] [0x6f3ff310] [debug]   [OpenAuto] [AndroidAutoEntity] destroy.

OryanBaker, 2018-12-30:
Phone shows connection, but still showing Waiting for Device

Pobyeda, 2019-01-05:
Try to run it like this:
sudo su
cd ~
cd openauto/bin
./autoapp

It helped to resolve the issue when OpenAuto is not acting when phone connected.

closed 2019-01-17 by f1xpl