f1xpl/openauto #153: deb/PPA please?

BloodyIron, 2018-12-30:
Any chance we can get deb packages and/or a PPA for this? Considering this seems to be intended for Rasp Pi, shouldn't be too hard to just make builds for that each time a new release comes out.

Please? :)

JMK19, 2019-04-19:
Hi,
If you want to install this simply and easily, use https://github.com/novaspirit/androidauto_rpi_install 
It's pretty simple:
```
git clone https://github.com/novaspirit/androidauto_rpi_install
cd androidauto_rpi_install
sudo chmod +x install.sh
./install.sh
```
It builds it all for you.

BloodyIron, 2019-04-19:
Yeah you're missing the point. With a deb or PPA the package manager can handle installation and uninstallation. I (and others) prefer to rely on the package manager, so we can keep track of what is (and isn't) installed.

JMK19, 2019-04-19:
I see.
I'm not a Linux expert but after analyzing a .rpm package it contains a .cpio package which contains folders that get copied to / (merging with other folders) so it's possible someone could build it, then put required folders into this sort of package
I'm analyzing a .deb right now.