f1xpl/openauto #157: Android Auto quits after few seconds (usb/wifi)

29axe, 2019-01-12:
I bought OpenAuto today but can't connect my phone correctly.
The phone connects and launches Android Auto but after a few seconds it turns back to OpenAuto interface (screen: "Connect to Android Auto").

I tried to use USB and Wi-Fi, both have the same result. 

Could it be related to the AndroidAuto version on the phone? (mine is 3.8.584564-release). App works correctly on phone. 
I have the same behavior using Crankshaft (openauto fork). It has been working fine for months but suddently stopped working few days ago.

tryharderpi, 2019-01-13:
I have the same issue with my Samsung S8 and I can add:

- a second phone (s7 edge) works 100% correctly with this latest crankshaft / openauto
- my S8 works correctly on my Sony ax100 Android AUto auto car unit
 -it's not the cable or usb port (as I tried 3 with Openauto and they all dó work on the Sony)
 - in the debug, after 10 seconds i can see the error:
**[error]   [OpenAuto] [AndroidAutoEntity] ping timer exceeded** and then **[OpenAuto] [App] onAndroidAutoQuit.**

Thanks for the good work, this project is awesome :)

29axe, 2019-01-13:
@tryharderpi I'd be curious to know which version of android is installed on both your phones. And also which android auto version.

tryharderpi, 2019-01-13:
@29axe Android 8 on both the phones.

Same Android Auto version for both: v3.9.585054-release (I also tried 3.8.584553)

On the S8+ there's Google Play services v 15.0.85 (040400-228608434)
On the S7Edge it's Google Play services v 14.7.99 (040408-223214910)

29axe, 2019-01-13:
I looked at the log too. 
This is the same error : 

> [error] [OpenAuto] [AndroidAutoEntity] ping timer exceeded.

[openauto.log](https://github.com/f1xpl/openauto/files/2753183/openauto.log)

f1xpl, 2019-01-13:
Ping timer exceeded means that phone does not send any data to OpenAuto for more than 10 seconds. Looks like the issue with the phone itself.

29axe, 2019-01-13:
Same for me. AA launches and responds correctly like before. Can also play songs, navigate.. but always after a few seconds it stops.

29axe, 2019-01-13:
Also I can mention that if I let the phone connected to the rpi I get an error on the phone after some time.
"Android auto encountered an error. Communication error 12. Something went wrong. Try to unplug and plug again the phone when you can."

[image]

29axe, 2019-01-13:
@tryharderpi
I finally managed to make it work. 

I noticed I was running a beta version of Google Play Services (v 15.xxx, same as yours). 
I left the beta program for this app in Play Store.
Then as it is an administrator app, I could not reinstall it to replace with non-beta version so I backed up all my data and reset my phone to factory. 

After setting everything back up I have now this configuration:
Android Auto 3.9.585054
Google Play Services 14.7.99

AA is now working correctly (except those annoying transitions between apps that are really slow on rpi and stop in the middle but I'll deal with it for now).

Thank you for your help. Hope you'll manage to make it work too. ;-)

PS: I'm now worried about the future stable version of Google Play Services. :/

tryharderpi, 2019-01-14:
Haha, very cool! I suspected it as well, but I was looking for a way to downgrade. I'll leave it for now, wait for the next openauto. It's still weird my old Sony HU works with my phone, even with that newer Google Play Services. Thanks for the trouble, have fun with the project :)

29axe, 2019-02-05:
As I mentionned few weeks back, the beta version  (v. 15.x) of Google Play Services made the android auto app crash. Reverting (to v. 14.x) did fix the problem. The issue is that this beta version is now released and with the latest update of my Samsung S9+ open auto is now unusable. Does anyone encountered this issue too?

tetofonta, 2019-02-05:
Yup, just yesterday I tried openauto with my Galaxy S9 without success.
I made it working by removing the ping routine (i suppose) in src/autoapp/service/AndroidAutoEntity.cpp, here's git diff report:
```
diff --git a/src/autoapp/Service/AndroidAutoEntity.cpp b/src/autoapp/Service/AndroidAutoEntity.cpp
index 50aabf8..31e63e2 100644
--- a/src/autoapp/Service/AndroidAutoEntity.cpp
+++ b/src/autoapp/Service/AndroidAutoEntity.cpp
@@ -249,7 +249,7 @@ void AndroidAutoEntity::triggerQuit()
 
 void AndroidAutoEntity::schedulePing()
 {
-    auto promise = IPinger::Promise::defer(strand_);
+/*    auto promise = IPinger::Promise::defer(strand_);
     promise->then([this, self = this->shared_from_this()]() {
         this->sendPing();
         this->schedulePing();
@@ -263,7 +263,7 @@ void AndroidAutoEntity::schedulePing()
         }
     });
 
-    pinger_->ping(std::move(promise));
+    pinger_->ping(std::move(promise));*/
 }
```

As you can see, commenting the entire function may make the whole thing working, but I think it's way off target.

yusijs, 2019-02-05:
It's the same for me (Mate 10 pro). Hopefully @f1xpl will be able to sort it somehow - Openauto pro started crashing on me this morning, and now tried crankshaft with same issues.

f1xpl, 2019-02-05:
@tetofonta Lack of ping functionality will break the wireless feature and may lead to random disconnections. Unfortunately it looks like reverse engineering is needed to fix this issue.

@yusijs OpenAuto Pro 3.1 uses totally different Android Auto backend. If you faced any issue with the Pro version just try to update to the latest version.

cydia2020, 2019-02-06:
Can ANYONE confirm whether AA works with OpenAuto Pro 3.1 or not? Will purchase if it works.

cydia2020, 2019-02-06:
@29axe confirmed working with Pixel XL.

inspector71, 2019-02-07:
I think it's important to realise that Play Services is Google's way of owning your phone. It underpins everything. 

If you don't want too many experiences like this in the future, best not to opt into being Google Human Testbots.

Let the people who get paid by the big G do the testing! They're a mega billion dollars company. They can afford to test software much more extensively.

You don't have to do their job for them, especially when there's often nothing more than "Big fixes, performance improvements" for release notes. 

For all we know their bug fixes could be nothing more than severe edge cases irrelevant to the majority and yet there's changes rolled into every release that could CAUSE worse bugs than they fix.

We're almost constantly indoctrinated with the newer is better dogma but it's not always the case!

Kman91, 2019-02-07:
This is a pretty serious issue.  Lots of people using Crankshaft are down, including me.  I used to use this everyday and it worked great until yesterday, now more and more people are down.
https://www.reddit.com/r/crankshaft/comments/ant3xt/crashing_shortly_after_i_plug_in_my_phone/ 
https://github.com/opencardev/crankshaft/issues/291

cydia2020, 2019-02-07:
@kman91 don't delay, buy OpenAuto Pro today.

Kman91, 2019-02-07:
I may do that, although $25 seems kind of steep for something I haven't even tried.  I'm all for supporting programmers.  I just prefer to support them after I know the solution is well done.

cydia2020, 2019-02-07:
@Kman91 
I don't usually shill for closed source softwares, however in this situation, I have to say that as someone who uses Android Auto daily but not able to fit a third party headunit to my car (cough SUBARU), crankshaft is too janky for my taste (random disconnections, ugly UI, slow startup, settings not saving, et cetera). OpenAuto Pro boots up a LOT faster despite having more features and a more polished interface. However prior to you making the purchase, I should advise you that unlike crankshaft, OpenAuto Pro requires an external soundcard/mic to function, if you don't have **either** of them, OpenAuto Pro will refuse to work, which means a few more dollars to purchase a mic from AliExpress. Also, OpenAuto Pro does not support light sensors natively, nor does it support auto switching between day/night based on RTC time.

So purchase wisely.

yusijs, 2019-02-07:
It has the same features as Crankshaft and more (bluetooth handsfree, dab radio support ++), but does not have sensors built in (that is to say, you need to add them manually).

Kman91, 2019-02-07:
I agree about many of the points you raise.  It's good to know that it's more stable (although I haven't had any issues with this until now) and quicker to start.  I just bought an Audio Injector Zero as a sound card.  I hope that works.  I haven't checked into supported sound cards for Openauto.

Not supporting light sensors is a huge minus for me, although it's nice that it makes brightness easy-ish to control.  Being blinded by my screen when I turn on the car isn't cool.  It's hard to believe a solution like this doesn't support a sensor.  I also can't believe neither solution supports automatic night/day based on phone time (which is always right), but that may be an AA problem.

I'll look into it.  Thanks for the info, it is very helpful!

yusijs, 2019-02-07:
It supports the sensors - you just have to add them yourself, instead of via the GUI like in CS

cydia2020, 2019-02-07:
@yusijs any instructions or guides would be appreciated.

yusijs, 2019-02-07:
Just google "raspbian add <sensorname>". Adding e.g an rtc is very simple (like 2 steps) :-) 

OA pro is built on top of Raspbian, so you have full access to the entire OS. So it's def more possibilities for sensors, but there are a bit more steps involved :)

cydia2020, 2019-02-07:
RTC is fine, I've done it a long time ago while foundling with the Pi 2B, the thing is to integrate the light sensor with OpenAuto, I probably know how to write the bash file but the issue remains whether OpenAuto can accept night/day switching commands from an external application.

GavinAshford, 2019-02-07:
I made the same change that @tetofonta mentions to comment out the contents of the scheduledPing method and OpenAuto works again for me with the latest Google Play Services (v15.0.90). It was stable on my 40minute commute home today while using Google Maps navigation as well as streaming BBC iPlayer Radio. I have a Pi 3B+, Pixel 2XL and original Google A-C cable.

picard001, 2019-02-08:
Same issue here, LG G7 running Google Play Services 15.0.9

ThePiLocator, 2019-02-08:
Same issue here, Samsung Galaxy S8 Google Play Services 15.0.9

29axe, 2019-02-09:
Updating Open Auto Pro to 3.1 fixed the issue for me. But I would prefer to use crankshaft. I'll wait for the next update from open auto.

ugifractal, 2019-02-10:
same issue happen on my nokia 6.1 phone.

bruschetta85, 2019-02-14:
hello guys, but what do you know will be corrected even openauto the open souce version?
I have almost completed a project that I have been working for about 1 year and I find myself openauto unusable. if openauto pro has been corrected because it does not provide the correction also for the open souce one.
I did several tests to fix the file AndroidAutoEntity.cpp but I think there are other problems because I do not get the desired result.

Kman91, 2019-02-15:
I upgraded to Pro and it's working now.  I like it, although it still needs a light sensor input and day/night switching based on cell phone time.  It does boot faster than Crankshaft.

Thanks Cydia!

bruschetta85, 2019-02-15:
here is what I feared, given the problems everyone will go pro and the open source project will be abandoned.
For me it is essential to restore the operation of openauto free because it is a fundamental component for the project I'm doing. I have been working for 1 year and I do not want to abandon the whole project for the play service 15.0.90
[https://onedrive.live.com/?authkey=%21AIFmb9xBsqMGN68&id=E3E1807ACE0DACC%211169&cid=0E3E1807ACE0DACC](url)

yusijs, 2019-02-15:
@bruschetta85 you can always fork and try fixing. While I would also prefer @f1xpl maintaining OA, open source is not sustainable, and I can fully understand that he's prioritizing OA pro. I'm assuming he has a job / other things in his life also consuming his time, so his priorities are most likely personal life > open auto pro > open auto opensource :-)

caipivara, 2019-02-19:
@yusijs this issue is fixed on OpenAuto pro?

yusijs, 2019-02-19:
Yes

Oper92, 2019-03-17:
Adding request.set_timestamp(1); fixed this issue for me 
```
void AndroidAutoEntity::sendPing()
{
    auto promise = aasdk::channel::SendPromise::defer(strand_);
    promise->then([]() {}, std::bind(&AndroidAutoEntity::onChannelError, this->shared_from_this(), std::placeholders::_1));
    aasdk::proto::messages::PingRequest request; 
    request.set_timestamp(1);
    controlServiceChannel_->sendPingRequest(request, std::move(promise));
}
```