f1xpl/openauto #158: Blank buttons on Waiting for device screen...

jonlowe, 2019-01-16:
Compiled successfully on Raspbian Stretch. Application opens, but the four buttons on the right side are blank, white with no lettering, highlighted or not. Currently using the RPi HDMI port on a TV, no touchscreen. I was able to find a screenshot of that screen and figured out what the buttons were, and activated the mouse pointer. Buttons seem to work ok, although tabs on Settings screen are faint.

ugifractal, 2019-01-16:
I have experienced same issue few months ago, and fixed by using sudo, when running the app.

jonlowe, 2019-01-17:
Thanks, hadn't thought of that. It's been awhile since I've used Linux. I will give it a try tomorrow. Still needs to be fixed.

jonlowe, 2019-01-17:
This can be combined with #106 and closed. I completely missed that open issue yesterday when I wrote this.

closed 2019-01-17 by jonlowe