f1xpl/openauto #162: After device is connected and AA is shown it goes back to the waiting for device screen

JaspervRijbroek, 2019-02-06:
So I have followed all the steps in the Wiki,
Compilations went without any issues

When I connect my device the android auto app starts, this will work for a few seconds,
after that no matter what I do I get reverted back to thw "Waiting for device" screen, without any issues logged.

Also when I log pulseaudio it shows that a channel is freed up, unfortunately without any reason why.

Could anyone please help me with this issue?

`
DISPLAY=:0 /home/pi/openauto/bin/autoapp 
libEGL warning: DRI2: failed to authenticate
Fontconfig warning: ignoring UTF-8: not a valid region tag
[2019-02-06 18:15:29.487195] [0x723c4000] [warning] [OpenAuto] [RecentAddressesList] failed to read configuration file: openauto_wifi_recent.ini, error: openauto_wifi_recent.ini: cannot open file. Empty list will be used.
[2019-02-06 18:15:29.535337] [0x6f9ff310] [info]    [OpenAuto] [App] Waiting for device...
[2019-02-06 18:15:29.565438] [0x6e1ff310] [info]    [OpenAuto] [App] Devices enumeration result: true
[2019-02-06 18:15:30.177986] [0x6e9ff310] [info]    [OpenAuto] [App] Device connected.
[2019-02-06 18:15:30.213769] [0x723c4000] [debug]   [OpenAuto] [AudioInput] create.
[2019-02-06 18:15:30.256862] [0x6e9ff310] [info]    [OpenAuto] [AndroidAutoEntity] start.
[2019-02-06 18:15:30.256983] [0x6e9ff310] [info]    [OpenAuto] [AudioInputService] start.
[2019-02-06 18:15:30.257581] [0x6e9ff310] [info]    [OpenAuto] [AudioService] start, channel: SYSTEM_AUDIO
[2019-02-06 18:15:30.257708] [0x6e9ff310] [info]    [OpenAuto] [SensorService] start.
[2019-02-06 18:15:30.257829] [0x6e9ff310] [info]    [OpenAuto] [VideoService] start.
[2019-02-06 18:15:30.257932] [0x6e9ff310] [info]    [OpenAuto] [BluetoothService] start.
[2019-02-06 18:15:30.258030] [0x6e9ff310] [info]    [OpenAuto] [InputService] start.
[2019-02-06 18:15:30.383285] [0x6e9ff310] [info]    [OpenAuto] [AndroidAutoEntity] version response, version: 1.5, status: 0
[2019-02-06 18:15:30.383409] [0x6e9ff310] [info]    [OpenAuto] [AndroidAutoEntity] Begin handshake.
[2019-02-06 18:15:30.415936] [0x6e9ff310] [info]    [OpenAuto] [AndroidAutoEntity] Handshake, size: 2282
[2019-02-06 18:15:30.491622] [0x6e9ff310] [info]    [OpenAuto] [AndroidAutoEntity] continue handshake.
[2019-02-06 18:15:30.502493] [0x6e9ff310] [info]    [OpenAuto] [AndroidAutoEntity] Handshake, size: 51
[2019-02-06 18:15:30.503234] [0x6e9ff310] [info]    [OpenAuto] [AndroidAutoEntity] Auth completed.
[2019-02-06 18:15:30.522253] [0x6e9ff310] [info]    [OpenAuto] [AndroidAutoEntity] Discovery request, device name: Android, brand: LGE Nexus 5X
[2019-02-06 18:15:30.522364] [0x6e9ff310] [info]    [OpenAuto] [AudioInputService] fill features.
[2019-02-06 18:15:30.522453] [0x6e9ff310] [info]    [OpenAuto] [AudioService] fill features, channel: SYSTEM_AUDIO
[2019-02-06 18:15:30.522497] [0x6e9ff310] [info]    [OpenAuto] [SensorService] fill features.
[2019-02-06 18:15:30.522544] [0x6e9ff310] [info]    [OpenAuto] [VideoService] fill features.
[2019-02-06 18:15:30.522610] [0x6e9ff310] [info]    [OpenAuto] [BluetoothService] fill features
[2019-02-06 18:15:30.522662] [0x6e9ff310] [info]    [OpenAuto] [InputService] fill features.
[2019-02-06 18:15:30.639651] [0x6e9ff310] [info]    [OpenAuto] [AndroidAutoEntity] requested audio focus, type: 4
[2019-02-06 18:15:30.639731] [0x6e9ff310] [info]    [OpenAuto] [AndroidAutoEntity] audio focus state: 3
[2019-02-06 18:15:30.641944] [0x6e9ff310] [info]    [OpenAuto] [AudioInputService] open request, priority: 0
[2019-02-06 18:15:30.642002] [0x6e9ff310] [info]    [OpenAuto] [AudioInputService] open status: 0
[2019-02-06 18:15:30.644065] [0x6f9ff310] [info]    [OpenAuto] [AudioService] open request, channel: SYSTEM_AUDIO, priority: 0
[2019-02-06 18:15:30.644444] [0x6f9ff310] [debug]   [OpenAuto] [AudioService] channel: SYSTEM_AUDIO audio output sample rate: 16000, sample size: 16, channel count: 1
[2019-02-06 18:15:30.646769] [0x6e1ff310] [info]    [OpenAuto] [VideoService] open request, priority: 0
[2019-02-06 18:15:30.646842] [0x6e1ff310] [info]    [OpenAuto] [OMXVideoOutput] open.
[2019-02-06 18:15:30.647796] [0x6e9ff310] [info]    [OpenAuto] [SensorService] open request, priority: 0
[2019-02-06 18:15:30.647917] [0x6e9ff310] [info]    [OpenAuto] [SensorService] open status: 0
[2019-02-06 18:15:30.648097] [0x6f1ff310] [info]    [OpenAuto] [InputService] open request, priority: 0
[2019-02-06 18:15:30.648148] [0x6f1ff310] [info]    [OpenAuto] [InputService] open status: 0
[2019-02-06 18:15:30.650712] [0x6f1ff310] [info]    [OpenAuto] [SensorService] sensor start request, type: 13
[2019-02-06 18:15:30.652515] [0x6e9ff310] [info]    [OpenAuto] [InputService] binding request, scan codes count: 0
[2019-02-06 18:15:30.652688] [0x6e9ff310] [info]    [OpenAuto] [InputDevice] start.
[2019-02-06 18:15:30.652773] [0x6e9ff310] [info]    [OpenAuto] [InputService] binding request, status: 0
[2019-02-06 18:15:30.654326] [0x6e9ff310] [info]    [OpenAuto] [SensorService] sensor start request, type: 10
[2019-02-06 18:15:30.662044] [0x6e1ff310] [info]    [OpenAuto] [VideoService] open status: 0
[2019-02-06 18:15:30.663434] [0x6e9ff310] [info]    [OpenAuto] [VideoService] setup request, config index: 3
[2019-02-06 18:15:30.663502] [0x6e9ff310] [info]    [OpenAuto] [OMXVideoOutput] init, state: true
[2019-02-06 18:15:30.663736] [0x6e9ff310] [info]    [OpenAuto] [VideoService] setup status: 2
[2019-02-06 18:15:30.664113] [0x6e1ff310] [info]    [OpenAuto] [VideoService] video focus indication.
[2019-02-06 18:15:30.668931] [0x6e1ff310] [info]    [OpenAuto] [VideoService] start indication, session: 0
[2019-02-06 18:15:30.687855] [0x6f9ff310] [info]    [OpenAuto] [AudioService] open status: 0, channel: SYSTEM_AUDIO
[2019-02-06 18:15:30.690039] [0x6f9ff310] [info]    [OpenAuto] [AudioService] setup request, channel: SYSTEM_AUDIO, config index: 1
[2019-02-06 18:15:30.690115] [0x6f9ff310] [info]    [OpenAuto] [AudioService] setup status: 2, channel: SYSTEM_AUDIO
[2019-02-06 18:15:47.716983] [0x6f9ff310] [error]   [OpenAuto] [AndroidAutoEntity] ping timer exceeded.
[2019-02-06 18:15:47.717107] [0x6f9ff310] [info]    [OpenAuto] [App] quit.
[2019-02-06 18:15:47.717173] [0x6f9ff310] [info]    [OpenAuto] [AndroidAutoEntity] stop.
[2019-02-06 18:15:47.717236] [0x6f9ff310] [info]    [OpenAuto] [AudioInputService] stop.
[2019-02-06 18:15:47.717530] [0x6f9ff310] [info]    [OpenAuto] [AudioService] stop, channel: SYSTEM_AUDIO
[2019-02-06 18:15:47.720249] [0x6f9ff310] [info]    [OpenAuto] [SensorService] stop.
[2019-02-06 18:15:47.720387] [0x6f9ff310] [info]    [OpenAuto] [VideoService] stop.
[2019-02-06 18:15:47.720453] [0x6f9ff310] [info]    [OpenAuto] [OMXVideoOutput] stop.
[2019-02-06 18:15:47.732839] [0x6f9ff310] [info]    [OpenAuto] [BluetoothService] stop.
[2019-02-06 18:15:47.732964] [0x6f9ff310] [info]    [OpenAuto] [InputService] stop.
[2019-02-06 18:15:47.733026] [0x6f9ff310] [info]    [OpenAuto] [InputDevice] stop.
[2019-02-06 18:15:47.734781] [0x6f9ff310] [info]    [OpenAuto] [App] Waiting for device...
[2019-02-06 18:15:47.735199] [0x6e9ff310] [error]   [OpenAuto] [VideoService] channel error: AaSdk error code: 30, native code: 0
[2019-02-06 18:15:47.735255] [0x6f9ff310] [error]   [OpenAuto] [SensorService] channel error: AaSdk error code: 30, native code: 0
[2019-02-06 18:15:47.735322] [0x6e1ff310] [error]   [OpenAuto] [AudioInputService] channel error: AaSdk error code: 30, native code: 0
[2019-02-06 18:15:47.735383] [0x6e9ff310] [error]   [OpenAuto] [SensorService] channel error: AaSdk error code: 30, native code: 0
[2019-02-06 18:15:47.735441] [0x6f9ff310] [error]   [OpenAuto] [AndroidAutoEntity] channel error: AaSdk error code: 30, native code: 0
[2019-02-06 18:15:47.735493] [0x6e1ff310] [error]   [OpenAuto] [BluetoothService] channel error: AaSdk error code: 30, native code: 0
[2019-02-06 18:15:47.735594] [0x6f1ff310] [error]   [OpenAuto] [AudioService] channel error: AaSdk error code: 30, native code: 0, channel: SYSTEM_AUDIO
[2019-02-06 18:15:47.735647] [0x6f9ff310] [debug]   [OpenAuto] [AndroidAutoEntity] destroy.
`

mvlanga, 2019-02-06:
+1 - Got the same issue, I think this problem appeared after the last android auto update

JaspervRijbroek, 2019-02-06:
I think it's something to do with udev because the initial data is set like the weather.

AndyCap1, 2019-02-06:
Same issue here, from reading another thread (https://github.com/f1xpl/openauto/issues/157) it appears it's something to do with "Google Play Services v 15.xxx" 

Mine was working fine until Google Play services updated on my phone.

Fernando-olm, 2019-05-14:
> I think it's something to do with udev because the initial data is set like the weather.

I also think that now (after the ping timeout correction) the problem is in permissions or udev. If I start my OpenAuto directly from the ~/openauto/bin/autoapp it runs for a couple minutes (seems to be random) and then it stops working. But when I do sudo su and start the same binary as before, it runs for hours without a problem. I'm using Stretch Raspbian with a Raspberry Pi 3B+.

overthrowrobotics, 2019-05-26:
Raspberry Pi 3B+ running Raspbian downloaded yesterday. I did a git clone, compiled, connected my Pixel 3 XL and after 5 seconds of showing Android Auto on screen it goes back to the connect your phone (but my phone still shows Android Auto is connected). I tried the udev rule and rebooted, same result. 5 seconds. Tried the change in #170 same result. Every time I see AaSdk error code: 30 when I exit. Seems I'm not alone and am running a really standard setup and this has been going on for a long time. Is this project dead?

Oper92, 2019-05-26:
#170 is still working for me, Samsung Galaxy S8 with Google Play Services v17.1, please check again if you correctly applied it, have you compiled again after have applied the patch? If you did everything right then please paste here your full log error with your Google Play Services version so that someone can try to help.

IkerJau, 2019-05-29:
> Raspberry Pi 3B+ running Raspbian downloaded yesterday. I did a git clone, compiled, connected my Pixel 3 XL and after 5 seconds of showing Android Auto on screen it goes back to the connect your [...]

I had the same problem. Tried #170 but didn't work to me. Then I commented the code lines shown at #157 and the app worked fine. I used it for 30 mins, more or less, and it didn't go back to waiting device screen, so it may be solved.

Fernando-olm, 2019-05-29:
Yep, I was too early in my review!
On the first day I had my phone connected to the car trough my home's network, in Wireless mode, and I let it there for a couple hours just checking it at random times. I guess it worked because I was doing nothing in Android Auto interface: no music playing, no map display, no "Ok Google" commands... 
After some days using it without a problem via USB (in smaller intervals, like 20 ~ 30 min), I needed to use it for more than 40 minutes, and then it stopped. When it was safe for me to check my phone, the red screen saying that Android Auto had a problem appeared. So I can agree that the problem stills there, but now it is getting longer to happen.