f1xpl/openauto #166: Interrupting connection after a few seconds

vince999, 2019-02-16:
Hello,

half a year ago I decided with a colleague in his car to install a raspberry with openauto. At that time we just set up the raspberry and everything went perfect.
However, this project went a bit in the dark. Than he recently got a new car and we wanted to finish this project.

The problem is now, as soon as i connect the smartphone to the raspberry, Android Auto opens but after about 10 seconds, it simply breaks down the connection.

Now when i start Openauto on a terminal, plug and unplug the phone, in the terminal i get:
libEGL warning: DRI2: failed to authenticate
qt5ct: using qt5ct plugin
[2019-02-16 21:13:42.752210] [0x72490000] [warning] [OpenAuto] [Configuration] failed to read configuration file: openauto.ini, error: openauto.ini: cannot open file. Using default configuration.
[2019-02-16 21:13:42.802135] [0x72490000] [warning] [OpenAuto] [RecentAddressesList] failed to read configuration file: openauto_wifi_recent.ini, error: openauto_wifi_recent.ini: cannot open file. Empty list will be used.
[2019-02-16 21:13:42.820376] [0x6faff310] [info]    [OpenAuto] [App] Waiting for device...
inotify_add_watch("/home/pi/.config/qt5ct") failed: "No such file or directory"
[2019-02-16 21:13:43.823906] [0x6f2ff310] [info]    [OpenAuto] [App] Devices enumeration result: false
[2019-02-16 21:14:02.334628] [0x6eaff310] [info]    [OpenAuto] [App] Device connected.
[2019-02-16 21:14:02.348969] [0x72490000] [debug]   [OpenAuto] [AudioInput] create.
[2019-02-16 21:14:02.359821] [0x6eaff310] [info]    [OpenAuto] [AndroidAutoEntity] start.
[2019-02-16 21:14:02.359897] [0x6eaff310] [info]    [OpenAuto] [AudioInputService] start.
[2019-02-16 21:14:02.360217] [0x6eaff310] [info]    [OpenAuto] [AudioService] start, channel: MEDIA_AUDIO
[2019-02-16 21:14:02.360271] [0x6eaff310] [info]    [OpenAuto] [AudioService] start, channel: SPEECH_AUDIO
[2019-02-16 21:14:02.360311] [0x6eaff310] [info]    [OpenAuto] [AudioService] start, channel: SYSTEM_AUDIO
[2019-02-16 21:14:02.360351] [0x6eaff310] [info]    [OpenAuto] [SensorService] start.
[2019-02-16 21:14:02.360399] [0x6eaff310] [info]    [OpenAuto] [VideoService] start.
[2019-02-16 21:14:02.360443] [0x6eaff310] [info]    [OpenAuto] [BluetoothService] start.
[2019-02-16 21:14:02.360493] [0x6eaff310] [info]    [OpenAuto] [InputService] start.
[2019-02-16 21:14:02.391050] [0x6faff310] [info]    [OpenAuto] [AndroidAutoEntity] version response, version: 1.5, status: 0
[2019-02-16 21:14:02.391139] [0x6faff310] [info]    [OpenAuto] [AndroidAutoEntity] Begin handshake.
[2019-02-16 21:14:02.400575] [0x6eaff310] [info]    [OpenAuto] [AndroidAutoEntity] Handshake, size: 2282
[2019-02-16 21:14:02.442652] [0x6eaff310] [info]    [OpenAuto] [AndroidAutoEntity] continue handshake.
[2019-02-16 21:14:02.449614] [0x6e2ff310] [info]    [OpenAuto] [AndroidAutoEntity] Handshake, size: 51
[2019-02-16 21:14:02.450109] [0x6e2ff310] [info]    [OpenAuto] [AndroidAutoEntity] Auth completed.
[2019-02-16 21:14:02.461435] [0x6e2ff310] [info]    [OpenAuto] [AndroidAutoEntity] Discovery request, device name: Android, brand: OnePlus ONEPLUS A5000
[2019-02-16 21:14:02.461570] [0x6e2ff310] [info]    [OpenAuto] [AudioInputService] fill features.
[2019-02-16 21:14:02.461672] [0x6e2ff310] [info]    [OpenAuto] [AudioService] fill features, channel: MEDIA_AUDIO
[2019-02-16 21:14:02.461719] [0x6e2ff310] [info]    [OpenAuto] [AudioService] fill features, channel: SPEECH_AUDIO
[2019-02-16 21:14:02.461753] [0x6e2ff310] [info]    [OpenAuto] [AudioService] fill features, channel: SYSTEM_AUDIO
[2019-02-16 21:14:02.461786] [0x6e2ff310] [info]    [OpenAuto] [SensorService] fill features.
[2019-02-16 21:14:02.461833] [0x6e2ff310] [info]    [OpenAuto] [VideoService] fill features.
[2019-02-16 21:14:02.461877] [0x6e2ff310] [info]    [OpenAuto] [BluetoothService] fill features
[2019-02-16 21:14:02.461907] [0x6e2ff310] [info]    [OpenAuto] [InputService] fill features.
[2019-02-16 21:14:02.492661] [0x6eaff310] [info]    [OpenAuto] [AndroidAutoEntity] requested audio focus, type: 4
[2019-02-16 21:14:02.492770] [0x6eaff310] [info]    [OpenAuto] [AndroidAutoEntity] audio focus state: 3
[2019-02-16 21:14:02.495056] [0x6e2ff310] [info]    [OpenAuto] [AudioInputService] open request, priority: 0
[2019-02-16 21:14:02.495245] [0x6e2ff310] [info]    [OpenAuto] [AudioInputService] open status: 0
[2019-02-16 21:14:02.495592] [0x6faff310] [info]    [OpenAuto] [AudioService] open request, channel: MEDIA_AUDIO, priority: 0
[2019-02-16 21:14:02.495662] [0x6faff310] [debug]   [OpenAuto] [AudioService] channel: MEDIA_AUDIO audio output sample rate: 48000, sample size: 16, channel count: 2
[2019-02-16 21:14:02.496440] [0x6e2ff310] [info]    [OpenAuto] [AudioService] open request, channel: SPEECH_AUDIO, priority: 0
[2019-02-16 21:14:02.496605] [0x6e2ff310] [debug]   [OpenAuto] [AudioService] channel: SPEECH_AUDIO audio output sample rate: 16000, sample size: 16, channel count: 1
[2019-02-16 21:14:02.498427] [0x6f2ff310] [info]    [OpenAuto] [AudioService] open request, channel: SYSTEM_AUDIO, priority: 0
[2019-02-16 21:14:02.498527] [0x6f2ff310] [debug]   [OpenAuto] [AudioService] channel: SYSTEM_AUDIO audio output sample rate: 16000, sample size: 16, channel count: 1
[2019-02-16 21:14:02.499341] [0x6eaff310] [info]    [OpenAuto] [SensorService] open request, priority: 0
[2019-02-16 21:14:02.499414] [0x6eaff310] [info]    [OpenAuto] [SensorService] open status: 0
[2019-02-16 21:14:02.500077] [0x6eaff310] [info]    [OpenAuto] [VideoService] open request, priority: 0
[2019-02-16 21:14:02.500148] [0x6eaff310] [info]    [OpenAuto] [OMXVideoOutput] open.
[2019-02-16 21:14:02.508961] [0x6faff310] [info]    [OpenAuto] [AudioService] open status: 0, channel: MEDIA_AUDIO
[2019-02-16 21:14:02.509710] [0x6faff310] [info]    [OpenAuto] [InputService] open request, priority: 0
[2019-02-16 21:14:02.509788] [0x6faff310] [info]    [OpenAuto] [InputService] open status: 0
[2019-02-16 21:14:02.510339] [0x6faff310] [info]    [OpenAuto] [SensorService] sensor start request, type: 13
[2019-02-16 21:14:02.510959] [0x6faff310] [info]    [OpenAuto] [AudioService] setup request, channel: MEDIA_AUDIO, config index: 1
[2019-02-16 21:14:02.511041] [0x6faff310] [info]    [OpenAuto] [AudioService] setup status: 2, channel: MEDIA_AUDIO
[2019-02-16 21:14:02.511575] [0x6f2ff310] [info]    [OpenAuto] [AudioService] open status: 0, channel: SYSTEM_AUDIO
[2019-02-16 21:14:02.511767] [0x6f2ff310] [info]    [OpenAuto] [InputService] binding request, scan codes count: 0
[2019-02-16 21:14:02.511829] [0x6f2ff310] [info]    [OpenAuto] [InputDevice] start.
[2019-02-16 21:14:02.511907] [0x6f2ff310] [info]    [OpenAuto] [InputService] binding request, status: 0
[2019-02-16 21:14:02.512411] [0x6f2ff310] [info]    [OpenAuto] [SensorService] sensor start request, type: 10
[2019-02-16 21:14:02.514022] [0x6faff310] [info]    [OpenAuto] [AudioService] setup request, channel: SYSTEM_AUDIO, config index: 1
[2019-02-16 21:14:02.514105] [0x6faff310] [info]    [OpenAuto] [AudioService] setup status: 2, channel: SYSTEM_AUDIO

RtApiPulse::callbackEvent(): the stream is closed ... this shouldn't happen![2019-02-16 21:14:02.516309] [0x6e2ff310] [info]    [OpenAuto] [AudioService] open status: 0, channel: SPEECH_AUDIO
[... 32 lines cut ...]
[2019-02-16 21:14:12.389763] [0x6f2ff310] [debug]   [OpenAuto] [AndroidAutoEntity] destroy.

Then when i open the openauto.log, in short, i get:
[2019-02-16 20:44:57.715118] [0x6fa8e310] [error]   [OpenAuto] [AndroidAutoEntity] ping timer exceeded.
[2019-02-16 20:44:57.740727] [0x6ea8e310] [error]   [OpenAuto] [VideoService] channel error: AaSdk error code: 30, native code: 0
[2019-02-16 20:44:57.740794] [0x6ea8e310] [error]   [OpenAuto] [AudioService] channel error: AaSdk error code: 30, native code: 0, channel: SYSTEM_AUDIO
[2019-02-16 20:44:57.740842] [0x6ea8e310] [error]   [OpenAuto] [AudioService] channel error: AaSdk error code: 30, native code: 0, channel: SPEECH_AUDIO
[2019-02-16 20:44:57.740887] [0x6ea8e310] [error]   [OpenAuto] [SensorService] channel error: AaSdk error code: 30, native code: 0
[2019-02-16 20:44:57.740932] [0x6ea8e310] [error]   [OpenAuto] [SensorService] channel error: AaSdk error code: 30, native code: 0
[2019-02-16 20:44:57.740977] [0x6ea8e310] [error]   [OpenAuto] [AudioService] channel error: AaSdk error code: 30, native code: 0, channel: MEDIA_AUDIO
[... cut ...]
[2019-02-16 20:44:57.741064] [0x6ea8e310] [error]   [OpenAuto] [AudioInputService] channel error: AaSdk error code: 30, native code: 0
[2019-02-16 20:44:57.741226] [0x6fa8e310] [error]   [OpenAuto] [AndroidAutoEntity] channel error: AaSdk error code: 30, native code: 0

The rest are a bunch of infos and debug informations.

Here the complete file:
[openauto.log](https://github.com/f1xpl/openauto/files/2872332/openauto.log)

Things I already tried:
-reinstall raspbian with openauto
-reinstall android auto on phone 
-udev rules

I hope there is a solution for this and thanks in advance.

rizlas, 2019-02-17:
imho #157 Duplicate

PaoloRenz, 2019-04-17:
Hello, I have the same problem but don't know how to solve it. Please help us!