f1xpl/openauto #169: Better fix for Google Play Services v15 bug

jessanros, 2019-02-26:
This is  more solid way of fixing the issue where Android Auto turns back to OpenAuto interface after a few seconds. It still checks for errors and will timeout when it needs to, it will just ignore the false alarm that occurs 10 seconds after you plug in your phone.

During installation, after downloading openauto with the "git clone" command, and before running "mkdir openauto_build", run

`sudo nano openauto/src/autoapp/Service/AndroidAutoEntity.cpp`

(you might have to adjust if you're using a different openauto installation directory)

At the bottom, replace the function

```
void AndroidAutoEntity::schedulePing()
{
    auto promise = IPinger::Promise::defer(strand_);
    promise->then([this, self = this->shared_from_this()]() {
        this->sendPing();
        this->schedulePing();
    },
    [this, self = this->shared_from_this()](auto error) {
        if(error != aasdk::error::ErrorCode::OPERATION_ABORTED &&
           error != aasdk::error::ErrorCode::OPERATION_IN_PROGRESS)
        {
            OPENAUTO_LOG(error) << "[AndroidAutoEntity] ping timer exceeded.";
            this->triggerQuit();
        }
    });

    pinger_->ping(std::move(promise));
}
```

With

```
void AndroidAutoEntity::schedulePing()
{
    auto promise = IPinger::Promise::defer(strand_);
    promise->then([this, self = this->shared_from_this()]() {
        this->sendPing();
        this->schedulePing();
    },
    [this, self = this->shared_from_this()](auto error) {
        if(error == aasdk::error::ErrorCode::NONE)
        {
            OPENAUTO_LOG(info) << "[AndroidAutoEntity] Evaded Google Services 15 v15.xxx error";
        }
        else if(error != aasdk::error::ErrorCode::OPERATION_ABORTED &&
           error != aasdk::error::ErrorCode::OPERATION_IN_PROGRESS)
        {
            OPENAUTO_LOG(error) << "[AndroidAutoEntity] ping timer exceeded.";
            this->triggerQuit();
        }
    });

    pinger_->ping(std::move(promise));
}
```

Apparently OpenAuto doesn't recognize a command from the new Google Services update and classifies it as a "ping timer exceeded error"

Hope it helps!

Mohammed-sha, 2019-03-13:
Hi All,

I have compiled openauto for ubuntu 16.04 and run autoapp. When connected Mobile phone to PC its says Android auto encountered an error, communication error 2.

Tried on two versions of Google Play service 15.0.90 and 14.5.74(MI phones). I have added the above patch and tried. 

[autoapp_debug.txt](https://github.com/f1xpl/openauto/files/2959800/autoapp_debug.txt)

[image]

Mohammed-sha, 2019-03-15:
@parkerlreed  I have tried with the above changes as you have mentioned, but android auto screen is not coming, in Mobile phone android auto is not loading instead its getting crashed. Tried with two phones.

i Have attached the openauto logs.

[openauto_debug_with_fix.txt](https://github.com/f1xpl/openauto/files/2969524/openauto_debug_with_fix.txt)