f1xpl/openauto #171: Not getting proper android auto screen on board

Mohammed-sha, 2019-03-22:
Hi All,

I have cross-compiled openauto for my board, when i run autoapp i am not getting proper video output on the screen.( refer attached image file). When i enabled gstreamer debug with loglevel 3, it says "input stream is corrupt".  
Could someone help me on this ?

> ./autoapp

 

> [  260.367231] usb 3-1: USB disconnect, device number 5
> 
> [2019-03-19 07:10:34.116569] [0x0000ffff8f32e9f0] [warning] [OpenAuto] [RecentAddressesList] failed to read configuration file: openauto_wifi_recent.ini, error: opena
> uto_wifi_recent.ini: cannot open file. Empty list will be used.
> Using Wayland-EGL
> Using the 'xdg-shell-v6' shell integration
> [2019-03-19 07:10:34.169051] [0x0000ffff8cb23120] [info]    [OpenAuto] [App] Waiting for device...
> [2019-03-19 07:10:34.174439] [0x0000ffff877fe120] [info]    [OpenAuto] [App] Devices enumeration result: false
> [  266.676286] usb 3-1: new high-speed USB device number 6 using ci_hdrc
> [  267.848822] usb 3-1: USB disconnect, device number 6
> [  268.204300] usb 3-1: new high-speed USB device number 7 using ci_hdrc
> [2019-03-19 07:10:38.268181] [0x0000ffff877fe120] [info]    [OpenAuto] [App] Device connected.
> [2019-03-19 07:10:38.284092] [0x0000ffff8f32e9f0] [debug]   [OpenAuto] [AudioInput] create.
> [2019-03-19 07:10:38.311151] [0x0000ffff8f32e9f0] [debug]   [OpenAuto] [QtVideoOutput] create.
> [2019-03-19 07:10:38.451171] [0x0000ffff877fe120] [info]    [OpenAuto] [AndroidAutoEntity] start.
> [2019-03-19 07:10:38.451281] [0x0000ffff877fe120] [info]    [OpenAuto] [AudioInputService] start.
> [2019-03-19 07:10:38.451966] [0x0000ffff877fe120] [info]    [OpenAuto] [AudioService] start, channel: MEDIA_AUDIO
> [2019-03-19 07:10:38.452038] [0x0000ffff877fe120] [info]    [OpenAuto] [AudioService] start, channel: SPEECH_AUDIO
> [2019-03-19 07:10:38.452083] [0x0000ffff877fe120] [info]    [OpenAuto] [AudioService] start, channel: SYSTEM_AUDIO
> [2019-03-19 07:10:38.452125] [0x0000ffff877fe120] [info]    [OpenAuto] [SensorService] start.
> [2019-03-19 07:10:38.452171] [0x0000ffff877fe120] [info]    [OpenAuto] [VideoService] start.
> [2019-03-19 07:10:38.452227] [0x0000ffff877fe120] [info]    [OpenAuto] [BluetoothService] start.
> [2019-03-19 07:10:38.452279] [0x0000ffff877fe120] [info]    [OpenAuto] [InputService] start.
> [2019-03-19 07:10:38.453380] [0x0000ffff87fff120] [info]    [OpenAuto] [AndroidAutoEntity] version response, version: 1.5, status: 0
> [2019-03-19 07:10:38.453436] [0x0000ffff87fff120] [info]    [OpenAuto] [AndroidAutoEntity] Begin handshake.
> [2019-03-19 07:10:38.461410] [0x0000ffff877fe120] [info]    [OpenAuto] [AndroidAutoEntity] Handshake, size: 2311
> [2019-03-19 07:10:38.509011] [0x0000ffff877fe120] [info]    [OpenAuto] [AndroidAutoEntity] continue handshake.
> [2019-03-19 07:10:38.512897] [0x0000ffff86ffd120] [info]    [OpenAuto] [AndroidAutoEntity] Handshake, size: 51
> [2019-03-19 07:10:38.513295] [0x0000ffff86ffd120] [info]    [OpenAuto] [AndroidAutoEntity] Auth completed.
> [2019-03-19 07:10:38.522587] [0x0000ffff87fff120] [info]    [OpenAuto] [AndroidAutoEntity] Discovery request, device name: Android, brand: Xiaomi POCO F1
> [2019-03-19 07:10:38.522696] [0x0000ffff87fff120] [info]    [OpenAuto] [AudioInputService] fill features.
> [2019-03-19 07:10:38.522797] [0x0000ffff87fff120] [info]    [OpenAuto] [AudioService] fill features, channel: MEDIA_AUDIO
> [2019-03-19 07:10:38.522855] [0x0000ffff87fff120] [info]    [OpenAuto] [AudioService] fill features, channel: SPEECH_AUDIO
> [2019-03-19 07:10:38.522905] [0x0000ffff87fff120] [info]    [OpenAuto] [AudioService] fill features, channel: SYSTEM_AUDIO
> [2019-03-19 07:10:38.522951] [0x0000ffff87fff120] [info]    [OpenAuto] [SensorService] fill features.
> [2019-03-19 07:10:38.523015] [0x0000ffff87fff120] [info]    [OpenAuto] [VideoService] fill features.
> [2019-03-19 07:10:38.523072] [0x0000ffff87fff120] [info]    [OpenAuto] [BluetoothService] fill features
> [2019-03-19 07:10:38.523114] [0x0000ffff87fff120] [info]    [OpenAuto] [InputService] fill features.
> [2019-03-19 07:10:38.553093] [0x0000ffff877fe120] [info]    [OpenAuto] [AndroidAutoEntity] requested audio focus, type: 4
> [2019-03-19 07:10:38.553169] [0x0000ffff877fe120] [info]    [OpenAuto] [AndroidAutoEntity] audio focus state: 3
> [2019-03-19 07:10:38.553940] [0x0000ffff877fe120] [info]    [OpenAuto] [AudioInputService] open request, priority: 0
> [2019-03-19 07:10:38.554013] [0x0000ffff877fe120] [info]    [OpenAuto] [AudioInputService] open status: 0
> [2019-03-19 07:10:38.554402] [0x0000ffff877fe120] [info]    [OpenAuto] [AudioService] open request, channel: MEDIA_AUDIO, priority: 0
> [2019-03-19 07:10:38.554471] [0x0000ffff877fe120] [debug]   [OpenAuto] [AudioService] channel: MEDIA_AUDIO audio output sample rate: 48000, sample size: 16, channel c
> ount: 2
> [2019-03-19 07:10:38.555055] [0x0000ffff87fff120] [info]    [OpenAuto] [AudioService] open request, channel: SPEECH_AUDIO, priority: 0
> [2019-03-19 07:10:38.555127] [0x0000ffff87fff120] [debug]   [OpenAuto] [AudioService] channel: SPEECH_AUDIO audio output sample rate: 16000, sample size: 16, channel 
> count: 1
> [2019-03-19 07:10:38.555791] [0x0000ffff8cb23120] [info]    [OpenAuto] [AudioService] open request, channel: SYSTEM_AUDIO, priority: 0
> [2019-03-19 07:10:38.555874] [0x0000ffff8cb23120] [debug]   [OpenAuto] [AudioService] channel: SYSTEM_AUDIO audio output sample rate: 16000, sample size: 16, channel 
> count: 1
> [2019-03-19 07:10:38.558261] [0x0000ffff86ffd120] [info]    [OpenAuto] [SensorService] open request, priority: 0
> [2019-03-19 07:10:38.558359] [0x0000ffff86ffd120] [info]    [OpenAuto] [SensorService] open status: 0
> [2019-03-19 07:10:38.559041] [0x0000ffff86ffd120] [info]    [OpenAuto] [VideoService] open request, priority: 0
> [2019-03-19 07:10:38.559161] [0x0000ffff86ffd120] [info]    [OpenAuto] [VideoService] open status: 0
> [2019-03-19 07:10:38.559745] [0x0000ffff86ffd120] [info]    [OpenAuto] [InputService] open request, priority: 0
> [2019-03-19 07:10:38.559814] [0x0000ffff86ffd120] [info]    [OpenAuto] [InputService] open status: 0
> [2019-03-19 07:10:38.561088] [0x0000ffff86ffd120] [info]    [OpenAuto] [SensorService] sensor start request, type: 13
> [2019-03-19 07:10:38.561939] [0x0000ffff86ffd120] [info]    [OpenAuto] [VideoService] setup request, config index: 3
> [2019-03-19 07:10:38.562124] [0x0000ffff86ffd120] [info]    [OpenAuto] [VideoService] setup status: 2
> [2019-03-19 07:10:38.563129] [0x0000ffff86ffd120] [info]    [OpenAuto] [InputService] binding request, scan codes count: 0
[... 27 lines cut ...]
> 
> 0:00:01.957501000  4787     0x11a33680 WARN                    v4l2 gstv4l2object.c:1986:gst_v4l2_object_get_interlace_mode: Driver bug detected - check driver with v
> 4l2-compliance from http://git.linuxtv.org/v4l-utils.git
> 
> 0:00:01.957806375  4787     0x11a33680 WARN                    v4l2 gstv4l2object.c:1986:gst_v4l2_object_get_interlace_mode: Driver bug detected - check driver with v
> 4l2-compliance from http://git.linuxtv.org/v4l-utils.git
> 
> 0:00:01.957849375  4787     0x11a33680 WARN                    v4l2 gstv4l2object.c:1986:gst_v4l2_object_get_interlace_mode: Driver bug detected - check driver with v
> 4l2-compliance from http://git.linuxtv.org/v4l-utils.git
> 
> 0:00:01.994370500  4787 0xfffed41feca0 WARN                    v4l2 gstv4l2object.c:1986:gst_v4l2_object_get_interlace_mode: Driver bug detected - check driver with v
> 4l2-compliance from http://git.linuxtv.org/v4l-utils.git
> 
> 0:00:01.994474000  4787 0xfffed41feca0 WARN                    v4l2 gstv4l2object.c:1986:gst_v4l2_object_get_interlace_mode: Driver bug detected - check driver with v
> 4l2-compliance from http://git.linuxtv.org/v4l-utils.git
> 
> 0:00:02.079181250  4787 0xfffed41feca0 WARN            v4l2videodec gstv4l2videodec.c:870:gst_v4l2_video_dec_decide_allocation:<v4l2h264dec0> Duration invalid, not se
> tting latency
> 0:00:02.326257875  4787     0x115eba00 FIXME                    bin gstbin.c:4336:gst_bin_query: implement duration caching in GstBin again
> 0:00:02.466636500  4787     0x11a33680 WARN               h264parse gsth264parse.c:1196:gst_h264_parse_handle_frame:<h264parse0> input stream is corrupt; it contains 
> a NAL unit of length 0
> 0:00:02.467172875  4787     0x11a33680 WARN               h264parse gsth264parse.c:1196:gst_h264_parse_handle_frame:<h264parse0> input stream is corrupt; it contains 
> a NAL unit of length 0
> 
[image]

Oper92, 2019-03-23:
This could be a phone specific issue, googling your device followed by Android Auto leads to many people with the same problem as yours. Have you tried to run OpenAuto with another phone?

Mohammed-sha, 2019-03-23:
@oper92,
Thanks for the reply.

I don't think the issue is with mobile phone as the openauto was working well when I connected Mobile phone (same phone) with ubuntu PC. the issue is coming when I connect with my reference board. I doubt it's the  issue with gstreamer package. Anyway I will try with another phone and update.