f1xpl/openauto #172: could not find qt5 and rtAudio

SpreadHateNotLove, 2019-03-27:
So, while trying to install OpenAuto on step six (instructions from wiki) I get this message, not sure how to solve it:

pi@raspberrypi:~/openauto_build $ cmake -DCMAKE_BUILD_TYPE=Release -DRPI3_BUILD=TRUE -DAASDK_INCLUDE_DIRS="/home/pi/aasdk/include" -DAASDK_LIBRARIES="/home/pi/aasdk/lib/libaasdk.so" -DAASDK_PROTO_INCLUDE_DIRS="/home/pi/aasdk_build" -DAASDK_PROTO_LIBRARIES="/home/pi/aasdk/lib/libaasdk_proto.so" ../openauto
-- Boost version: 1.62.0
-- Found the following Boost libraries:
--   system
--   log
--   unit_test_framework
--   date_time
--   log_setup
--   filesystem
--   thread
--   regex
--   chrono
--   atomic
-- Found libusb-1.0:
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib/arm-linux-gnueabihf/libusb-1.0.so
CMake Warning at CMakeLists.txt:32 (find_package):
  By not providing "FindQt5.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Qt5", but
  CMake did not find one.

  Could not find a package configuration file provided by "Qt5" with any of
  the following names:

    Qt5Config.cmake
    qt5-config.cmake

  Add the installation prefix of "Qt5" to CMAKE_PREFIX_PATH or set "Qt5_DIR"
  to a directory containing one of the above files.  If "Qt5" provides a
  separate development package or SDK, be sure it has been installed.

CMake Error at cmake_modules/Findrtaudio.cmake:64 (message):
  Could not find rtaudio
Call Stack (most recent call first):
  CMakeLists.txt:35 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/pi/openauto_build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/openauto_build/CMakeFiles/CMakeError.log".

parkerlreed, 2019-03-28:
Did you run the apt-get from that same page? It includes qt5 and rtaudio

```
sudo apt-get install -y libqt5multimedia5 libqt5multimedia5-plugins libqt5multimediawidgets5 qtmultimedia5-dev libqt5bluetooth5 libqt5bluetooth5-bin qtconnectivity5-dev pulseaudio librtaudio-dev librtaudio5a
```

SpreadHateNotLove, 2019-03-29:
I did run every single command without changing anything. I think I will run this second time and see where it gets me.

IkerJau, 2019-05-22:
> I did run every single command without changing anything. I think I will run this second time and see where it gets me.

I have the same problem. Did you solve it?