f1xpl/openauto #174: Dependencies

haywirephoenix, 2019-04-07:
For openauto and aasdk. Which dependencies are required to be kept after the application has been built? I'm working with buildroot on a minimal build.

parkerlreed, 2019-04-10:
If I played my cards right when making the Arch packages

OpenAuto
```
depends=('aasdk-git' 'qt5-connectivity' 'qt5-multimedia' 'pulseaudio' 'gst-libav' 'rtaudio')
```

aasdk
```
depends=('libusb' 'protobuf' 'openssl' 'boost-libs')
```

haywirephoenix, 2019-04-10:
Thanks for this mate.

Hey we have the same surname!

kooopah, 2019-05-13:
> For openauto and aasdk. Which dependencies are required to be kept after the application has been built? I'm working with buildroot on a minimal build.
> 
> P.s: you are a God for making this

First of all, I give you encouragement for this process xD the dependencies are basically those described by Openauto Wiki but ... using Buildroot require some other extra package. Enable in your configurator (menuconfig, nconfig, whatever ...) the Host-cmake package, will help you to build aasdk and openauto from crosscompiling and also the RTAUDIO dependency that you will have to build from source, always using all the flag -DCMAKE_TOOLCHAIN_FILE = / path / to / buildroot / output / host / usr / share / buildroot / toolchainfile.cmake + extras. Using buildroot with a minimal installation I got through some tweak boot time from 4 seconds to main with an rpi3b +, making the system a very good implementation for my car !!!

PD: if when running aoutoapp, QT returns some error related to the graphical environment, it adds the options to the boot -platform linuxfb: D