f1xpl/openauto #175: Using phone as a controller

NigerianTechno, 2019-04-14:
For the project I'm working on I'm not going to have a touch screen and all I'll be using OpenAuto for is to display Google Maps on the IMID in my car. So is there any way for me to use my phone to control open auto or at least somehow have it auto launch into maps?

Oper92, 2019-04-14:
You can fork the project and configure it to send an InputEventIndication after the VideoStream starts and a BindingRequest is received. It's a little tricky, because you should know prior the resolution that OpenAuto is going to use and find the exact pixel region, but this should simulate a click that should allow you to go in the Google Maps tab without interact with the screen.

NigerianTechno, 2019-04-14:
After thinking about it for a bit I realized that even if I managed to get it to go straight into maps I'd still have to find a way to use my phone keyboard every time I want to enter a new destination lol. Guess it's back to the drawing board, maybe I'll find a way to use my steering wheel controls or something.