f1xpl/openauto #176: OpenAuto loads for a short period of time, then quits

JMK19, 2019-04-19:
With a Samsung Galaxy S8, plugging in the phone with the app does load it, but after 30 or so seconds it goes back to the "waiting for device..." menu. Android 8.1
With a Google Pixel XL (the original) same thing happens, but after around 15 or so seconds it goes back to the "waiting for device..." menu. Android 9 Pie
Tried:

- Enabling USB debugging
- Google Pixel: set default USB mode to File Transfer
- Galaxy S8: set USB mode to MTP

I think it's that it's automatically resetting from File Transfer/MTP to Charge device (which is the new default for some reason...?) thus ending the signal. It does detect it as a car, though...

Oper92, 2019-04-19:
Check #170 https://github.com/f1xpl/openauto/pull/170/commits/f05074e26758be7edac1208c7c3e7eb3e7292605 this should fix your issues

JMK19, 2019-04-19:
Just tried that, using Novaspirit's openauto build script. Replaces f1xpl's OpenAuto to your branch. No luck.
Can you give me instructions on how to do this? Thanks.

racingbrett, 2019-05-01:
Did you try run it through the console and check the output, getting a similar issue and I believe its related to the line "ping timer exceeded", tried multiple fix suggestions and nothing seems to work.. Seems the paid for version of openauto has been fixed along with the crankshaft ng fork version but the open version of autoopen has been left to rot.

JMK19, 2019-05-04:
I use crankshaftNG now and it works fine, yes that's exactly my problem.

closed 2019-05-04 by JMK19