f1xpl/openauto #177: How to solve app crash after device connecting

dpnkrg, 2019-05-07:
How to solve this error, 

After connecting the device app is crashing,

here is the log 

> [2019-05-07 11:14:53.971711] [0xaf41c310] [info]    [OpenAuto] [InputService] stop.
[2019-05-07 11:14:53.972334] [0xaf41c310] [info]    [OpenAuto] [InputDevice] stop.
[2019-05-07 11:14:53.974172] [0xae41c310] [error]   [OpenAuto] [VideoService] channel error: AaSdk error code: 33, native code: 2
[2019-05-07 11:14:53.975276] [0xae41c310] [error]   [OpenAuto] [AudioService] channel error: AaSdk error code: 33, native code: 2, channel: SYSTEM_AUDIO
[2019-05-07 11:14:53.976904] [0xaec1c310] [error]   [OpenAuto] [SensorService] channel error: AaSdk error code: 33, native code: 2
[2019-05-07 11:14:53.977606] [0xaec1c310] [error]   [OpenAuto] [BluetoothService] channel error: AaSdk error code: 33, native code: 2
[2019-05-07 11:14:53.978292] [0xafc1c310] [error]   [OpenAuto] [SensorService] channel error: AaSdk error code: 33, native code: 2
[2019-05-07 11:14:53.978957] [0xafc1c310] [error]   [OpenAuto] [AudioInputService] channel error: AaSdk error code: 33, native code: 2
[2019-05-07 11:14:53.987923] [0xaf41c310] [info]    [OpenAuto] [App] Waiting for device...
[2019-05-07 11:14:53.996235] [0xaf41c310] [error]   [OpenAuto] [AndroidAutoEntity] channel error: AaSdk error code: 33, native code: 2
[2019-05-07 11:14:53.996567] [0xaf41c310] [debug]   [OpenAuto] [AndroidAutoEntity] destroy.

dpnkrg, 2019-05-07:
Facing error issue, error code 33 and error code 30.

Any help or suggestion?

Oper92, 2019-05-07:
Report the entire log error, but if you newly installed OpenAuto then you're probably facing the "ping timer exceed" error, check pull request #170 that modifies the AndroidAutoEntity file as shown here: https://github.com/f1xpl/openauto/pull/170/commits/f05074e26758be7edac1208c7c3e7eb3e7292605 then compile again and see if it works.
Probably related also to: #176 #157 #169

dpnkrg, 2019-05-08:
@Oper92,  

Fixed the error by the pull request #170 and modified the AndroidAutoEntity file.

Form next time, I will post the entire log error. 

Thanks for the help.

closed 2019-05-08 by dpnkrg