f1xpl/openauto #184: Please ADD the support for fm radio

mayukh1998, 2019-06-04:
Only thing missing in openauto pro  is the support for fm radio. Please add this. Every car stereo generally has a fm radio.

chrisfromwa, 2019-06-04:
They have a dedicated thread on their forum about this very topic.  I'm not sure if they are still looking into it,  or if they have decided to just drop it.  They haven't given any updates in awhile on it. 
https://bluewavestudio.io/forum/viewtopic.php?p=1903#p1903

But I think if more people started showing interest,  they might be motivated to make some progress on it.  So make sure to state your opinion on their forum and tell them you want FM radio support.  

It's something I need too,  I want to be able to use this software and not always have to use my internet connection from my phone.  To be a true Head Unit replacement,  it needs FM radio support.

A124Bob, 2019-06-06:
Also mind for DAB+ Radio!